guysoft/FullPageOS #7: When webpage is not in english chromium offers translation

jorgeiglesiasca, 2015-10-23:
If you open a non english web chromium opens a dialog covering the top part of the screen with options to translate.  Will be great to cancel that option by default as you dont have keyboard or mouse on screen.

tailorvj, 2016-06-02:
I would try adding  --disable-translate when running chromium from the command line

petrleocompel, 2017-11-23:
Also happened. Chrome detected my country and offered "CZ-cs" content translate to "EN".
Still not fixed.

guysoft, 2017-11-27:
Looks like [``--disable-translate`` was removed from chromium](https://chromium.googlesource.com/chromium/src/+/4baa4206fac22a91b3c76a429143fc061017f318).
So we can't disable it via commandline

However [here are instructions how to disable it from chromium](http://pearsonsupport.helpserve.com/Knowledgebase/Article/View/898/35/myenglishlab-turn-off-google-chrome-auto-translation-feature-when-using-myenglishlab).
Closing if anyone else has no input.

petrleocompel, 2017-11-27:
OK, but if you startup chrome in incognito mode it does not show up.
So i would recommend to add `--incognito`. Because i really dont have input on my screens.

guysoft, 2017-11-29:
I can't put ``--incognito`` as a default option to all users. You are welcome to put ``--incognito`` in your own installation.

petrleocompel, 2017-11-29:
I understand. Then would be nice to have some override to scripts from boot partition. Because do this on every installation is a lot of work. Or I missed something?

guysoft, 2017-11-29:
It might make sense to add a chromium.txt which would hold the chromium command execution settings.

petrleocompel, 2017-11-29:
So ok `/boot/chromium.txt` should hold command execution settings. (I'll make pull request later next week)

But maybe it will be much helpfull to add some first-time init script. 
Because today i discovered problem with different timezone on system.
It would be great have `/boot/(install|setup).sh` which will be executed to configure system.
Or maybe I'm just missing timezone configuration.

guysoft, 2017-11-29:
The timezone is a separate issue. We could create a timezone module for
CustomPiOS

marians, 2018-02-17:
Has anyone had success in suppressing the translate tool? I am running into this problem, too.

I've been looking for info on how to configure chromium so that it wouldn't show a translation tool. I found the language settings in this answer: https://unix.stackexchange.com/a/6223

Unfortunately I am lacking knowledge on how to apply that setting so that it actually sticks. Adding that "intl" key in `/home/pi/.config/chromium/Default/Preferences` had no effect for me. It was gone/overwritten after killing chromium. Would be glad for a hint here.

In addition I found https://cs.chromium.org/chromium/src/components/translate/core/browser/translate_prefs.cc which lists preferences keys that affect translation. Here `translate_blocked_languages` looks especially promising. But, same problem as above, I don't know how to inject that setting.

[Another answer](https://unix.stackexchange.com/a/218221) on the stackexchange page mentioned above suggests to install the `chromium-browser-l10n` package to localize chromium. However I couldn't test that, as the package couldn't be found by `apt-get`.

petrleocompel, 2018-02-18:
Put `—incognito` to chromium parameters.

guysoft, 2018-02-18:
Thanks for the input guys!
If you could tell me what needs to go where, I can add it to the build.
The ``—incognito`` solution is good for users, but I can't ship like that because then logins would not be stored.

petrleocompel, 2018-02-18:
I suggest it only as run option for running instance not to modify image.

I have customized script which I use to configure my instances. Easier than editing image.

guysoft, 2018-02-18:
I can edit the image.
Thats the point of this repo :)

petrleocompel, 2018-02-18:
You didn’t understand me. ☺️
I know you can. I know I don’t want to put there an --incognito option as default. But I’m trying to help that guy with translate popup. How to fix it on his running instance. 

I promised pull request for chromium options, but of course company in which I work didn’t give time to make it. We have only 12 running instances and I configured it only once so nobody touched that for a while. 

Once I will make another installation I will make pull request.

guysoft, 2018-02-18:
Ok, understood now :)

marians, 2018-02-18:
This is what seems to have solved the problem for me for now, for German web content:

- Installed `locales-all` package
- set the locale to `de_DE.UTF-8`
- Added `export LC_ALL=de_DE.UTF-8` to `home/pi/.profile`. Not sure if that was needed.
- rebooted

Still looking for a way to pre-configure this, ideally based on `FULLPAGEOS_OVERRIDE_LOCALE`.

marians, 2018-03-01:
I need some help with the change to apply the locale setting and make the Chromium translate tool disappear.

Here are my changes: https://github.com/netzbegruenung/FullPageOS/pull/1

As you can see [here](https://github.com/netzbegruenung/FullPageOS/pull/1/files#diff-402186ebd6c8a0e79ceb5a2e82a13e23R87) I am trying to modify `/home/pi/.profile`, but it doesn't work. In the final image, the file is untouched. Any idea what I would need to change?

guysoft, 2018-03-08:
Hey, just saw this now because you pull requested to your own repo and no to this repo.

1. can you provie a ``build.log``?
2. You can try adding your changes to ``end_chroot_script`` and not ``start_chroot_script`` the might be overwritten, but I am guessing without seeing the build.log.

marians, 2018-03-08:
@guysoft I am preparing the PR to the `devel` branch in our own forked repo first.

Thanks for the hint regarding `end_chroot_script`!

Where does `build.log` end up? Or are you talking about the terminal output?

guysoft, 2018-03-08:
``build.log`` is by default output in the path where you run the build command. If you are using vagrant thats the path of the src folder of your distro. is set at ``src/build`` (in CustomPiOS)
``[ -n "$LOG" ] || LOG="build.log"
``

christophe77, 2019-01-16:
The easyest solution is to add a meta to your html page : <meta name="google" content="notranslate">

jppicture, 2020-02-21:
in /home/pi/script/start_chromium_browser data put this parameter to startup "--disable-features=TranslateUI" and the message is away

closed 2020-02-27 by guysoft

Kevinsky86, 2022-02-03:
> in /home/pi/script/start_chromium_browser data put this parameter to startup "--disable-features=TranslateUI" and the message is away

This Git thread is Google's top search result when you look for this, so for the purpose general interest i'm posting a reply in this topic.
This variable has been renamed from TranslateUI to just Translate.

If /home/pi/script/start_chromium_browser data is adjusted accordingly and FullpageOS is subsequently restarted the translate bubble goes away.

commit bef14b91 (2020-02-27): Fix issue from #312, fixes #7