guysoft/FullPageOS #8: is it "Raspbian Compatible" ?

nicolasbd, 2015-10-28:
I mean, does this distro only run SSH & Chrome at startup, or is this still possible to install scripts, and why not quit chrome or run other applications in other sessions ?

thanks !

guysoft, 2015-10-28:
I am not sure what you mean by "raspbian compatible".
This disto is based on Raspbian. It has chrome and soon more tools around (see issue #5 for example).

In answer to your spesific questons:
1. You can quit chrome and run other applications, that sorta defies the point of the OS
2. Yes you can start a local web server and then point chrome to localhost.
3. You can edit the HTML on the Pi, but you will need to start some kind of tool, you will need to get the GUI to load something other than chrome, the startup script for X is here: 
https://github.com/guysoft/FullPageOS/blob/devel/src/filesystem/home/pi/scripts/run_onepageos
4. Yes you can install a webIDE on the pi and use chrome to edit it.

nicolasbd, 2015-10-29:
Yes, thanks, that's what I wanted to know !
"compatible" wasn't the right term, I was confused by the fact that you said it's based on a 3D printing oriented distro, and wanted to know if it was still possible do be used for normal stuff or if it was a special distro that would only contain specific packages for running a Chromium app at startup.
Thank you !

closed 2015-10-29 by guysoft