guysoft/FullPageOS #10: Wifi not working from fullpageos-network file

jorgeiglesiasca, 2015-10-29:
Tried to configure WIFI from the fullpageos-network file but does not work.  

If i go to X  server and configure from graphical screen it works and remains configurated for next reboots.

If i go to /etc/network/interfaces and edit there the same lines i have in fullpageos-network  it works. 

uncommented the lines IFACE - SSID - PASSWORD

linuxundich, 2015-11-03:
When you look closely at the boot messages you notice, that FullPageOS looks for a /boot/octopi-network.txt. When you copy /boot/fullpageos-network.txt to /boot/octopi-network.txt the wlan starts up -- i guess this has something to do with the fact that FullPageOS is built on Octopi.

guysoft, 2015-11-03:
@linuxundich Ok thats a good pointer, I located the problem!
Sould be fixed in the next version of FullPageOS.
In the meantime you can report if running this fixes it:

    sudo sed -i 's/octopi/fullpageos/g' /root/bin/map_iface

guysoft, 2015-11-23:
Update: you also need to change octopi to fullpageos in the fullpageos-network.txt file.

Has anyone got his to work so far?

closed 2015-12-22 by guysoft

guysoft, 2015-12-23:
Latest nightly fixes the issue, tested and confirmed:
http://docstech.net/FullPageOS/nightly/2015-12-22_2015-11-21-fullpageos-jessie-0.6.0.zip

commit 143ac590 (2015-12-22): Fix tupo that was missing, fixes #10