guysoft/FullPageOS #9: Not going full screen - No refresh

jorgeiglesiasca, 2015-10-29:
The page shown is maximized with only title bar and and the browser border visible.

Checked all the script files in the home folder and seems ok , tried to change kiosk mode to fullscreen mode in chromium but nothing happened. 

If i insert a keyboard and press F11 it goes correctly full screen.  Looks like the refreshing and F5 keystroke script written in the scripts folder are not working.

After some changes i installed again the untouched image in the SD card to see if something i´ve changed was forcing this issue but the problem seems to appear from scratch. On first boot the page goes maximized but not fullscreen.

 I couldnt check the refresh one after first boot because i have not internet connection.

guysoft, 2015-10-29:
1. Can you please send a photo?
2. Did you modify anything?
3. The script that makes chrome go fullscreen is at ``/home/pi/scripts/fullscreen`` , can you execute that over SSH and report the result? No other windows should be open at boot..

jorgeiglesiasca, 2015-10-29:
I made modifications but when i saw the border i wrote a new SD with fresh
system to check and browser border appears on first boot.

guysoft, 2015-11-29:
@jorgeiglesiasca There is a new nightly build that might solve this:
http://docstech.net/FullPageOS/nightly/

jorgeiglesiasca, 2015-12-04:
Tried nigthly but fullpageos-network file is missing in boot folder.   I added it manually , i inform just to add it in next releases.

guysoft, 2015-12-04:
@jorgeiglesiasca did fullpageos-network work for you after added? Did not work for me.

jorgeiglesiasca, 2015-12-04:
No. The system load diferent from before. No errors or warnings during
boot. After that a very long time of black screen and at the end raspbian
desktop.  No network working or browser load.

guysoft, 2015-12-29:
This should be fixed in the last nightly,

guysoft, 2016-04-03:
No response, tests show it works, and the refresh issue is addressed in #11

closed 2016-04-03 by guysoft