guysoft/FullPageOS #11: No Refresh - Feature Request

kf4mnc, 2015-12-28:
Hi,
I am really enjoying the OS you have put together for raspberry pi! One feature I would like to see added is the automatic refresh. I have been playing with it for sometime and not been in getting this to work. I know someone else mentioned this in October, but I wanted to see where it currently stands.

Thank you for putting together a great program!

guysoft, 2015-12-29:
The next version is to include a tab manager called FullPageDashbaord by @amitdar
Source is at: https://github.com/amitdar/FullPageDashboard

Adding refresh to it should not be hard, but we did not get around to it.

kf4mnc, 2015-12-31:
I have tried to add in the auto refresh using the refresh.sh file found in the scripts folder of the OS, however, it fails to run. When I send the following code via SSH it works fine. Any suggestion on what I am doing wrong?

I can send just this one line
DISPLAY=:0 xdotool getactivewindow key F5

Or I can send all of the lines below, both resulting in refreshing the screen.
export DISPLAY=":0"

WID=$(xdotool search --onlyvisible --class chromium|head -1)

xdotool windowactivate ${WID}

xdotool key ctrl+F5

Thank you for your help!

guysoft, 2016-01-02:
The script focuses the window and sends "F5", aka the refresh button. It can let you auto-refresh only if you are not using FullPageDashboard (given in the nightly build).
You need to make sure the active window is indeed chromium, if you opened terminals and other stuff it will not work.
Also, you can't just paste each line and it would work, because `WID=$(xdotool search --onlyvisible --class chromium|head -1)` is setting a variable, and variables in bash are not stored between command line executions, they are stored in scripts.

patricktokeeffe, 2019-01-09:
I'm using FullPageOS to display an Outlook calendar and needed autofresh functionality to ensure new events are displayed and the agenda automatically updates to the current day. (xref #35)

First I tried [disabling kiosk mode](https://github.com/guysoft/FullPageOS/issues/110#issuecomment-286263771) and installing the [Revolver-Tabs plugin](https://github.com/guysoft/FullPageOS/issues/94#issuecomment-323548717) but, unfortunately, tab rotation did not start automatically at launch (benhedrington/revolver-chrome-extensions/issues/17) so neither did the refreshing.

Instead, I had success adding a cron entry to run `~/scripts/safe_refresh` per @benyanke's [comment here](https://github.com/guysoft/FullPageOS/issues/127#issuecomment-358677257). Thanks! :+1:

stappersg, 2020-05-29:
Sixteen months later:  Please close this issue.

guysoft, 2020-05-29:
@stappersg 1. No its still a requested feature which might be solved.
2. Why are you asking to close an issue you are not even contributing to?

stappersg, 2020-05-30:
On Fri, May 29, 2020 at 01:10:30PM -0700, Guy Sheffer wrote:
> Why are you asking to close an issue you are not even contributing to?

Oops.

I did it again.

Probing the vitality of a project in a way
that is recieved as poking a project.

guysoft, 2020-05-31:
@stappersg 
Must say I have seen stranger comments.
But would recommend you should be more positive, PR some typo fix to the documentation. Otherwise comments like that might get misinterpreted as meddling. I have a few developer friends that zoned out all together form their projects because people kept poking them without any contribution to the project.