guysoft/FullPageOS #12: Nightly Build 12-27-15: Black Screen On Boot

NF3RN0, 2016-01-14:
I get a back screen with a cursor after the boot messages. I first tried changing the webpage in fullpageos.txt to no avail.

I can ssh to the Pi and can ping google from it so I know that I have internet connection to the pi.

If I kill Xorg and then try to restart the service from ./run_onepageos I get a list of several error messages that it cannot open the display.

xset:  unable to open display ""
xset:  unable to open display ""
xset:  unable to open display ""
matchbox: can't open display! check your DISPLAY variable.
14/01/2016 04:40:36 passing arg to libvncserver: -rfbauth
14/01/2016 04:40:36 passing arg to libvncserver: /home/pi/.vnc/passwd
14/01/2016 04:40:36 x11vnc version: 0.9.13 lastmod: 2011-08-10  pid: 953
14/01/2016 04:40:36 XOpenDisplay("") failed.
14/01/2016 04:40:36 Trying again with XAUTHLOCALHOSTNAME=localhost ...
14/01/2016 04:40:36 
14/01/2016 04:40:36 *** XOpenDisplay failed. No -display or DISPLAY.
14/01/2016 04:40:36 *** Trying ":0" in 4 seconds.  Press Ctrl-C to abort.
14/01/2016 04:40:36 *** 1 2 3 4 
14/01/2016 04:40:40 XOpenDisplay(":0") failed.
14/01/2016 04:40:40 Trying again with XAUTHLOCALHOSTNAME=localhost ...
14/01/2016 04:40:40 XOpenDisplay(":0") failed.
14/01/2016 04:40:40 Trying again with unset XAUTHLOCALHOSTNAME ...
14/01/2016 04:40:40 

14/01/2016 04:40:40 ***************************************
14/01/2016 04:40:40 *** XOpenDisplay failed (:0)

*** x11vnc was unable to open the X DISPLAY: ":0", it cannot continue.
*** There may be "Xlib:" error messages above with details about the failure.

This is with the latest nightly build, I'll try the october build to see if I can get that working.

I think it might be trivial to note that I am using a Pi Zero.

Cheers.

guysoft, 2016-01-31:
Hey,
There is a new nightly that might fix this issue. It waits for the an internal webserver to start before starting chromium.

jorgeiglesiasca, 2016-02-12:
In the nithtly from 31 of Jan  (lite version around 600mb)  take ages to load in Pi2  (around 10 minutes) and does not load in PiB 

If i unplug the pi when it hangs loading (happens allways in Pi B ) next boot gets into a loop and at the end i get a kernel panic. 

Tried to write 3 or 4 SD cards to check if was product of casuality and i always end in kernel panic.  This does not happen on Pi2

guysoft, 2016-02-13:
Hey, you wrote Pi2 twice, did you mean a different version on the first
mention?

samjrdn, 2016-02-29:
I am also seeing the OP's issue on a Pi B with the latest nightly.

tob1987, 2016-03-13:
Same issue here with all nighty builds, X11 starting with black screen and mouse but no chrome comes up, but has PID. Only stable version with wheezy and chromium 22 is working for me on Pi B+ v1 2. Any ideas?

guysoft, 2016-03-15:
It seems like everyone experiencing this has a Pi 1/B+/zero. It might be anything slower than a Pi2 is too slow for the recent distro.

luedm, 2016-03-24:
I'm using a Pi 3 and it is stuck on the colored screen when I boot. Unable to connect via SSH.

guysoft, 2016-03-24:
Regarding the Pi 3, I need to make a build using the new base image
released,  reports on OctoPi says that fixes the issue.

Regarding Pi 1/b+/Zero, I need to see, it might be booting times are
different. Or they have not enough power to run Chromium, or the Armv6 is
not behaving well with chromium build. I am not in a hurry to test all
these hypotheses because I am not sure how Chromium would behave on those
device, they are not likely to handle modern versions of Chromium. But
thanks for the reports, and reports on my guesses would be welcome.

guysoft, 2016-06-02:
Hey, if anyone wants to try the recent RC, it seems to work far more better and has a chance on the RaspberryPi 1/Zero/B+

bacacosta, 2016-06-03:
I tried the last nightly with a Raspberry Pi B and still have the black screen problem. Looks like this can't be fixed.

guysoft, 2016-06-15:
So the reason FullPageOS does not work on these devices ie since they use ARMv6, please see #43

guysoft, 2017-10-23:
Fixed, it works now, its slow as hell and takes 5-10 minutes to boot. Might work as a digital picture frame.

closed 2017-10-23 by guysoft