guysoft/FullPageOS #15: Nightly build 2016-03-14_2016-02-26-fullpageos-jessie-lite-0.6.0 Cursor not hiding

slindgren, 2016-03-15:
I've noticed that the Cursor doesn't seem to be hiding itself after loading web page.

guysoft, 2016-03-15:
I am not sure how to fix that, nor do I have the research time. A pull request would help.

slindgren, 2016-03-15:
It would be enough if the cursor would move away, or just move a bit sometimes. Then the Chromium + css should know how to hide it.. I think..

guysoft, 2016-04-03:
There was a pull request fixing this issue. Is this fixed in the nightly?

slindgren, 2016-04-03:
I have not downloaded the new nightly, but I'll get it tomorrow and check if everything is working!

slindgren, 2016-04-04:
now it requires login and the chromium isn't starting.. :/

guysoft, 2016-05-27:
I just tested, it still does not work, try latest nightly

tailorvj, 2016-06-02:
@guysoft I found this solution on AskUbuntu:
Try 'unclutter'
the purpose of this program is to hide the pointer after the mouse has not moved for a while. Using it, you can make the pointer appear only when the user touches the screen, and disappear right after it. (maybe this was not exactly what you were aiming for. But it is much easier than your alternative =P)
To use, install it
`sudo apt-get install unclutter`
then run it
`unclutter -idle 0.01 -root`
the number is a number of seconds before the pointer disappears (in this case, 1/100 seconds)

pricecomstock, 2016-06-04:
If you're getting "Could not open display" from unclutter, try the following.
`DISPLAY=:0.0 ; export DISPLAY
unclutter -idle 0.01 -root`

Hopefully that saves you a trip to [this thread.](https://www.reddit.com/r/Fedora/comments/1f58qf/remove_mouse_cursor_on_boot/)

andreicek, 2016-06-08:
If you want this done automatically on boot (or to be precise when X becomes available) you can add this to `/etc/X11/xinit/xinitrc`

So your file might look like this:
```
#!/bin/sh

# /etc/X11/xinit/xinitrc
#
# global xinitrc file, used by all X sessions started by xinit (startx)

# invoke global X session script
. /etc/X11/Xsession
unclutter -idle 0.01 -root
```

guysoft, 2016-06-16:
Likely this has been fixed by #34 , if not then we are going to add in 0.7.0 unclutter

guysoft, 2017-09-26:
We didn't add unclutter and this seems to be working.

closed 2017-09-26 by guysoft

G3z, 2018-02-16:
simply edited lightdm config in 
`/etc/lightdm/lightdm.conf`

changing line
`#xserver-command=X`
to
`xserver-command=X -nocursor`

guysoft, 2018-02-16:
Cool to know, but that would hide the cursor all together

slindgren, 2018-02-16:
Tätä sähköpostiosoitetta ei enää käytetä! Löydät uuden sähköpostiosoitteen Jumppahullun nettisivuilta!
  
 Ystävällisin terveisin
 Samuli Lindgren

G3z, 2018-02-16:
@guysoft yes but touch still works fine, it depends on what is the desired effect.

maxcel2, 2022-12-03:
I know that this is an old thread, but I'm still having a problem getting rid of the cursor in FullPageOS. G3z's solution of editing the lightdm.conf file doesn't do anything. 
Installing unclutter and running it from the command line brings up the cursor while the screen is touched, which then disappears quickly, as intended. However, if it is added to xinitrc, the cursor hangs around for about a second. 
I really want to get rid of the cursor altogether because the tip of the arrow is offset from the point of stylus contact and I have been unsuccessful so far in calibrating the screen.

milansimek, 2025-12-18:
In my case adding the `--cursor=none` flag to `/home/USER_NAME/scripts/start_chromium_browser` did the trick.