guysoft/FullPageOS #17: Nightly build does not work on RaspberryPi 3

guysoft, 2016-05-15:
Hey,
I tried FullPageOS image on raspberrypi 3 and it seems to boot in to the wrong GUI environment. Would be nice if anyone else could confirm this.

icthelp-spje, 2016-05-19:
Hey,

Confirmed here too. Boots to wrong environment, prompts me to login.

guysoft, 2016-05-19:
That is the only mirror I have that was kindly donated by docstech, I can sync to anywhere else if someone offers storage

icthelp-spje, 2016-05-20:
I'll contact the CECV and let them know it was incorrectly blocked. 

Will test an earlier build today and let you know.

matthova, 2016-05-25:
Confirmed.
Same issue for me

matthova, 2016-05-25:
Just tried with a Raspberry pi2 and had the same issue there as well. It's booting to the debian 8 login screen

closed 2016-05-27 by guysoft

guysoft, 2016-05-27:
Hey all,
The failure was caused because the build was only half built due to a regression, fixes now and tested on the recent nightly, get it at: http://docstech.net/FullPageOS/nightly/2016-05-27_2016-05-10-fullpageos-jessie-lite-0.6.0.zip
md5: 
http://docstech.net/FullPageOS/nightly/2016-05-27_2016-05-10-fullpageos-jessie-lite-0.6.0.zip.md5

icthelp-spje, 2016-05-31:
Unfortunately I couldn't wait for the issue to get fixed and just rolled my own solution. Thanks for resolving this though, will keep it in mind for the future. :)

guysoft, 2016-05-31:
@icthelp-spje sorry about that, you are always welcome to contribute your own solutions here.

commit 008f4912 (2016-05-27): Fix regression from new common.sh, fixes #17