guysoft/FullPageOS #18: FullPageOS 0.6.0 RC1

guysoft, 2016-05-29:
Hey,
The long delays FullPageOS 0.6 first release candidate, would be nice if people can confirm it works for them, or not:

Image:
http://docstech.net/FullPageOS/2016-05-27_2016-05-10-fullpageos-jessie-lite-0.6.0.zip
Md5:
http://docstech.net/FullPageOS/2016-05-27_2016-05-10-fullpageos-jessie-lite-0.6.0.zip.md5

guysoft, 2016-06-02:
Hey guys,
I can see there is a lot of traffic from [Hacker news](https://news.ycombinator.com/item?id=11809911. It would be nice if any of you that downloaded can report back here if this works well or bad for you. I am only one person and my QA relies on you.

guysoft, 2016-06-02:
Update from @tailorvj that the vnc password doea not work. Would be nice if anyone can confirm. If this is the case expect RC3

Dr-Juice, 2016-06-03:
I've installed it, and we're using it to display dashboards in our office.  Right now, we have a Pi 3 running FullPageOS to display the status page of our Observium environment.  I'm ordering 2 more Pi 3's to run as FullPageOS kiosks in other departments to display web pages from SQL reports in our material testing lab.

The issues I've found so far are, the VNX password isn't 'raspberry', and are there instructions on how to configure and use the FullPageDashboard app?

guysoft, 2016-06-04:
@Dr-Juice The VNC password issue is here, including a workaround #26 
There are no instructions for FullPageDashbaord, it was written by @amitdar

tailorvj, 2016-06-05:
@guysoft what's our schedule for 0.60 RC2 (or is it 0.61)?

guysoft, 2016-06-05:
I am going tonight to a Burning Man regional event for a week (Midburn)
after months of planning. So not untill that is over.
There is a build server, so I can release builds from my phone, but not
commit any code.
I suggest if someone wants FullPageOS to develop faster, then come to aid.
This is a project that I developed because I saw a need, and I think this
week with it going viral we saw its the case. But I can't test it and
maintain it myself.
I have my hands full with my startup and OctoPi already.
I will gladly teach people how to build it, and where to edit, its pretty
simple.

Dr-Juice, 2016-06-06:
Thanks for the reply, I'll give it a shot

guysoft, 2016-06-15:
RC2 which fixes the vnc password and mouse issues is now available at #44

closed 2016-06-15 by guysoft