guysoft/FullPageOS #19: Full page Electron?

vespakoen, 2016-05-31:
Hey! I think it would be cool if it could boot up an [electron](http://electron.atom.io/) app.

Would it be easy to do that?

guysoft, 2016-05-31:
You don't need to fork, you can make an electron variant in the variants folder, then build that. This would mean your build would always be up to date. See [Building FullPageOS variants section](https://github.com/guysoft/FullPageOS#building-fullpageos-variants)

As to setting it up, should not be a problem.
If you want to do it for yourself you can:

1. Install nodejs
2. Install electron
3. Install your electron app and make it start on boot
4. Point ``/boot/fullpageos.txt``to your app url

If you write a script for this you could make a variant of it

vespakoen, 2016-05-31:
Cool! didn't see that folder, I would like to do this but pretty busy at the moment, if nobody else will, I might in a (couple) of week(s)

heibor, 2016-06-03:
Nodejs may be a little heavy on the PI platform.

developit, 2016-06-03:
@heibor It's actually pretty decent!  Memory usage is obviously higher than other runtimes, but at long as you have 200M+ free, you'll see pretty reasonable performance.  Plus you can always set `--max-old-space-size` ([recommended](https://github.com/nodejs/node/issues/2738) for low-memory environments like the Pi) to configure when you want the garbage collector to kick in.

tailorvj, 2016-06-05:
I'm also interested in implementing node.js on FullPageOS. Let's create some kind of a virtual hackathon and give it a go. 
My aim is to replace the dashboard app with node.js and a chrome extension, as described on the following [thread](https://github.com/guysoft/FullPageOS/issues/24#issuecomment-223800053)

heibor, 2016-06-13:
I have a plan to reimplement dashboard with [openresty](http://www.openresty.org/), perhaps nginx+lua based solution can bring more performance and flexibility.

amicoleo, 2017-11-13:
Hei, any updates on this? I'd also like to have an Electron variant for FullPageOS.

guysoft, 2017-11-13:
No one picked it up. Also since there is a modules system now it should even be easier to build your own  OS. All you need is someone with the time (not me)

guysoft, 2019-03-09:
Hey,
So @tailorvj wrote an electron app, and I wrote a discro that starts it on boot. Tested and works, and solves this issue:
https://github.com/fullpageos/PiJam

Since 2016 when this issue was opened FullPageOS moved to use CustomPiOS system, which enables implementing this, but as a septate distro.

Closing, you are welcome to ask questions.

closed 2019-03-09 by guysoft