guysoft/FullPageOS #22: Graphics acceleration?

m2pc, 2016-06-02:
Hi, I'm curious if your OS supports any sort of graphics acceleration on the Pi? I recall trying to use Chromium on an older Pi and it was painfully slow to work with.  I also seem to remember some developers were working on a more modern X server that supported some of the acceleration available in the Pi's GPU or something similar.

Thanks and I'm excited to try your OS on some projects!

guysoft, 2016-06-02:
Indeed, its a pain. I know that the Pi3 has experimental acceleration stuff, but I didn't get time to understand how to implement it. See ``optional experimental GL desktop driver (can be enabled using advanced options in command-line raspi-config)`` mention in [raspbian release notes](http://downloads.raspberrypi.org/raspbian/release_notes.txt).
If anyone wants to test and report, I can add this.

heibor, 2016-06-02:
Improved performance is always exciting.

guysoft, 2016-12-27:
In RC1

closed 2016-12-27 by guysoft