guysoft/FullPageOS #23: Exit Chromium

TimDowker, 2016-06-02:
Is there a way to exit chromium once booting has completed?  I need to troubleshoot some networking issues and can't get back to the command line locally.

guysoft, 2016-06-02:
Good point,
1. You can run from tty (you enter tty1 by pressing doing ``ctrl+alt+F1`` ) ``DISPLAY=:0 xterm``
2. or you can just use the tty terminal there.
That should give you a terminal, exiting chromium usually results in a of restart of ``lightdm``, this is so it will keep running in case chromium crashed.
4. If you have a serial connection device you can use that too, [more information here](http://elinux.org/RPi_Serial_Connection)

If you need any other way to enter a terminal please share :)

guysoft, 2018-07-25:
Closing if no reply

closed 2021-12-23 by TimDowker

guysoft, 2021-12-23:
Thanks for cleaning up :)