guysoft/FullPageOS #24: Cannot access raspberrypi.local on 0.6 RC1

tailorvj, 2016-06-02:
When trying to access raspberrypi.local via vnc I didn't get an answer from the raspberry pi, so I looked up the DHCP host list on my router and it seems to recognize the FullPageOS raspberry pi as host name fullpageos. 
[image]
I actually tried to access it as well without success, so I tried the IP address given by DHCP and it worked. 
Another problem I found is the VNC password isn't set to raspberry. I had to SSH into the pi and run 
`/home/pi/scripts/setX11vncPass raspberry`
[image]

When accessing the pi via HTTP I got the default Apache landing page instead of the dashboard
[image]

soler2000, 2016-06-02:
I also have the same issues,  cannot reach the dashboard app at all

tailorvj, 2016-06-02:
@soler2000 do you mean via HTTP? Do you see the dashboard when the Pi is connected via HDMI to the screen? Did you try adding URLs to fullpagedashboard.txt on the SD card?

soler2000, 2016-06-02:
I could not connect via VNC until I added the password

How should the URLs be formatted / separated?   1 URL per line,  should they included http://

tailorvj, 2016-06-02:
My most recent test says editing fullpagedashboard.txt has no effect on the dashboard app. Changes made via VNC or http://fullpageos.local/FullPageDashboard/ are saved between restarts though. Personally I prefer the text file method better and I think the dashboard app should be rebuilt

mpiet, 2016-06-03:
I did some digging and the root directory for lighttpd is pointing to `/var/www/html`. However, FullPageDashboard is a sub-directory of the `html` folder. I changed the lighttpd conf in `/etc/lighttpd/lighttpd.conf` and changed `server.documentroot` to `/var/www/html/FullPageDashboard.` I then restarted the lighttpd service, but I guess you could also just reboot the pi, and everything started working.

tailorvj, 2016-06-03:
Thanks @mpiet this will be taken care of in the next release. You are more than welcome to fork and send @guysoft a pull request. 

Thinking a little further down the road, I'd like to replace the dashboard with another solution, probably node+express+websocket based. Once this is done we'll integrate it as variant probably

guysoft, 2016-06-03:
There is  slight mix in this issue, so let me split it in to topics
Regarding the router:
This is not a FullPageOS issue, and there are similar talks on OctoPi. For example see https://github.com/guysoft/OctoPi/issues/56 .

Lightttpd default page:
Indeed, that is the page you see at root. There is ``/welcome`` and ``/FullPageDashboard`` that hold the shipped apps. I thought leaving root free for now.

Regarding FullPageDashbaord:
@tailorvj You are welcome to suggest an alternative, at the moment FullPageDashboard is the best thing we have.

Reguarding the vnc password:
That might be an issue, I suggest we open a separate one for it. If so it would mean RC2.

@mpiet Changing the root to FullPageDashbaord would mean the welcome app would not work, and it means users can't put their own apps in root.

vanz541, 2016-06-03:
"_Regarding the router:
This is not a FullPageOS issue..._"
I believe part of the issue is that the FullPageDashboard welcome demo indicates that you should connect using VNC to "raspberrypi.local" to manage the tabs.  However, this is not the case.  The device is called "fullpageos.local" by default, so the instructions are incorrect as it relates to this image.  Not being able to connect by hostname when it is correct is a different story.

Also, I too was unable to access using VNC with a password of 'raspberry' as indicated, "Ships with preconfigured X11VNC, for remote connection (password 'raspberry')" on https://github.com/guysoft/FullPageOS  
Only after I set the password to raspberry as mentioned in the first comment was I able to connect.

tailorvj, 2016-06-05:
@guysoft thanks for [fixing the .local URL issue on the welcome page](https://github.com/tailorvj/FullPageOSWelcome/commit/94de8308aec8e5b278574c17149a96ba3274b317). 
I've completed the research necessary for replacing the dashboard app with a local node.js + chrome extension solution. I'll give it a shot in the near future. If there's any js coders around who'd like to help out in the implementation, please contact me. I can code it on my own, but it usually takes much more time because I'm a product manager and streaming expert, not a hardcore JS coder :) 
Regarding web server root, I think we should have redirection on index.php to a management app url defined on /boot along the other settings files.

guysoft, 2016-06-16:
Can I close this?

tailorvj, 2016-06-16:
I suggest we redirect lighttpd home to /welcome and include a link to the dashboard in the welcome page. 

Dashboard will be replaced in v 0.7.0 with node-based solution

Kind regards,
Tailor

closed 2016-06-30 by guysoft