guysoft/FullPageOS #29: Choppy Video

jacobgraf, 2016-06-04:
We are looking to use this for digital signage and occasionally we will have a short promo video play. We are using this as a test video (https://graftechnology.com/video/) but the video is really choppy. I set the GPU_MEM to 448 since that's the max for a 1GB Raspberry Pi 3 but it's still choppy. Any ideas how to get things running smoother? Thanks!

guysoft, 2016-06-04:
Currently chromium is not utilizing the GPU. To date I have not be able to find how to enable it. Talks are at #22 . Solutions are welcome.

tailorvj, 2016-06-05:
@jacobgraf could you please try [this out](http://www.webupd8.org/2014/01/enable-hardware-acceleration-in-chrome.html) and tell us if it solves the problem for you?

guysoft, 2016-06-05:
I am not sure that would solve it alone.

tcaddy, 2016-06-07:
I was able to get hardware decoding working from Chromium by enabling the experimental gl graphics driver in `raspi-config` and enabling the `--ignore-gpu-blacklist` chrome flag, but elements of videos still played choppy.

I tried another Raspberry Pi distribution (OpenElec / XBMC) that runs videos in a dedicated media player.  The same MP4 videos played smoothly in the media player app.

tailorvj, 2016-06-07:
Maybe we are trying to squeeze too much out of a web browser. Did you try smaller resolutions?

tcaddy, 2016-06-07:
@tailorvj smaller resolutions played smoother, but don't look that great.  In the end, I used a more powerful device for the video playback use case.

pricecomstock, 2016-06-26:
@jacobgraf Not in browser, but I'm having success using omxplayer. It goes fullscreen over the browser, and when the video is done, it closes and returns to the browser.

guysoft, 2016-12-27:
Hey, try RC1, it seems to be able to run fast, but it crashes, not sure why.

guysoft, 2016-12-29:
@jacobgraf Hey, if still relevant, it seems like webm does work on full screen at the moment, its mp4 container that seems to have a bug now. it means that if you convert your video to webm it should work with the new 0.7.0 RC1. Here is a webm video for reference:
http://video.webmfiles.org/big-buck-bunny_trailer.webm

Enzarro, 2017-05-15:
This dude managed to get hardware decoding in Chromium:
http://thebugfreeblog.blogspot.com/
It would be interesting if you take a look at it. Would be really useful to have a merge of these two projects.

guysoft, 2017-05-15:
@Enzarro that uses specilized code, we already have a better solution which you can download and use in RC1  #92

mark-veenstra, 2017-05-17:
I am trying to run a WebRTC site in FullPageOS. But as soon as I receive video it gets choppy/lagged instantly. You can try it by loading https://appr.tc as default page and connect the Pi and your PC to same room. Tested with RC1. Any ideas for optimilization?

ruvenss, 2017-05-19:
@mark-veenstra What's the hardware behind?

mark-veenstra, 2017-05-19:
Sorry forgot to mention. I used the `Pi 3B` and a normal HDMI screen which is used for computer also. Added an USB camera to the `Pi 3B` to enable cam feature. Used this cam on laptop also and worked fine.
The video is shown normal on laptop from the Pi, but all received video data on the Pi is choppy. CPU is also at max all the time.

Boscop, 2017-09-07:
Any update on this?
Isn't `rpi-chromium-mods` using hardware acceleration?

guysoft, 2017-09-11:
@Boscop It is, but it disables x264 hardware decoding which can be used to play Youtube videos.
See: https://github.com/guysoft/FullPageOS/issues/86#issuecomment-312369762

guysoft, 2017-09-11:
There might be progress on this:
```2017-09-07:
  * Bug fix for crash on some videos and animations in Chromium
```

Boscop, 2017-09-11:
It would be very useful to be able to play hardware accelerated H.264 in FullPageOS..

guysoft, 2017-09-11:
@boscop it can be done, as I already said, see https://github.com/guysoft/FullPageOS/issues/86#issuecomment-312369762

stzoran1, 2019-10-10:
I am trying to run mp4 video trough HTML5 video tag but it is very choppy. I have Raspberry PI 4 (2GB RAM) and tried almost every combination to fix it but without success. 

Any recommendations? I am ready to donate if proper solution is offered as my project depends on it.

guysoft, 2019-10-10:
@stzoran1 Can you provide a link to the webpage running the video of an equivalent? I can't debug it without an example I can test

stzoran1, 2019-10-10:
Thank you for quick reply. Here is the test link:
https://radio.numera.rs/rpidevices/12456/dekordom0710

There is a slide show containing one image and one video. When I use images only everything works perfectly, but video is choppy.

tcaddy, 2019-10-13:
It's been years since I dealt with this but if I remember correctly, the MP4 decoding chip on the Raspberry Pi is not open source.  Raspbian ships with a binary blob that's licensed from the company that provides the chip.  There's a media player that ships with Raspbian named "omxplayer" that will in fact play videos smoothly.  But this program won't hook into Chrome or Chromium nicely.

I think I tried some experimental flags in Chrome to enable unsupported GPUs.  I don't remember if it just didn't work or if it worked in some flakey manner that wasn't good enough.

I ended up not using the Raspberry Pi for the project at hand at the time.  (But I found a use for in another project.)

I hope things have improved since and you can get your goal accomplished.

pi43r, 2020-08-29:
That's strange. I have no hardware acceleration on my RPI 3B whatever I try.

Running the `enable_gpu` script quietly does nothing. And as far as I understand this is only for the GUI, right?

I activated fake and full KMS in `raspi-config`.

In the end I updated chromium and all other packages and followed this [guide](https://raspberrypi.stackexchange.com/questions/111642/raspberry-pi-4-framerate-drop-with-video-on-chromium-browser), which also didn't change anything for me. 

I have the feeling this should work in 2020, what am I doing wrong?

Here is a screenshot of what `chrome://gpu` gives me
[image]

guysoft, 2020-08-30:
@pi43r What version of FullPageOS?
0.11.0 Has version 74 of chrome and this is 78 - is this the nightly?

Also that guide has extra chromeium flags, did you try them?
```
-use-gl=egl --enable-gpu-rasterization --enable-native-gpu-memory-buffers --ignore-gpu-blacklist
```

pi43r, 2020-09-01:
Yes, the screenshot was after all my attempts to fix it.
I have tried both the nightly and stable version.

I was trying to run a small videoloop in the browser. For this project I have reverted to omxplayer with less interactivity, but I'm sure the same problem will come up in the future.
So, just to make sure: You have full hardware acceleration on your RPi in chromium?

ilyai, 2022-05-10:
I managed to get hardware acceleration working on FullPageOS 0.12.0 with the following Chromium flags:

    --enable-gpu-rasterization --enable-accelerated-video-decode --enable-native-gpu-memory-buffers --ignore-gpu-blocklist

Notice on flags mentioned previously: When using `--use-gl=egl` option it fails with `Failed to load GLES library: libGLESv2.so.2` error, and `--ignore-gpu-blacklist` is deprecated now.

@guysoft Just wondering if there's a way to set Chromium flags by changing some file on SD card, without booting, like WiFi configuration for example?

guysoft, 2022-05-10:
The file that holds all the flags is here:
https://github.com/guysoft/FullPageOS/blob/devel/src/modules/fullpageos/filesystem/home/pi/scripts/start_chromium_browser#L4