guysoft/FullPageOS #30: add fullpageos-webroot-redirect-url.txt settings file

tailorvj, 2016-06-05:
Currently, when accessing http://fullpageos.local the default lighttpd page is displayed in the browser. For the sake of HTTP management of the device, or even a simple info page, I think it would be great if we add another settings file on the SD card that defines the URL to redirect from web root. 
index.php should be updated to read the parameter from /boot/fullpageos-webroot-redirect-url.txt and redirect the browser.

patricktokeeffe, 2019-01-09:
This issue helped me discover the `/welcome` page *and* tipped me off that `/FullPageDashboard` isn't accessible because sometimes Chrome bites, not bc of a network issue. Thanks @tailorvj

guysoft, 2019-01-10:
Pull requests are welcome.