guysoft/FullPageOS #31: Websites Not Loading

rharkonen, 2016-06-06:
I am having a issue where when I try to add another tab it does not seem to load the page, it is just blank. I have tried it with multiple websites and for some reason only one website has been able to load. I can however connect through VNC with no problems and also hit F1 and log into a website that way. The site I am trying to get working is local on the network and I am adding the http:// in front of it so I know that is not the problem. Any idea on what would be causing this issue?

Thanks

tailorvj, 2016-06-06:
@rharkonen Are you adding the sites via the built-in dashboard app? Are the "internal" websites connected to the same network as the Pi? Did you try restarting the Pi? Please provide as much info as possible on your issue

Kind regards,
Tailor

rharkonen, 2016-06-06:
Yes I am adding it via the built-in dashboard. There are several internal website and I can connect to one of them no problem, but not to the other one. Same happens on external websites.. I have rebooted and even tried re-imaging the Raspberry Pi 3 a few times and nothing works. 

The first time I imaged it I had no problems, but then it crashed after awhile and I had to re-imaged it and that is when I started having the issues. Some sites work like https://www.hpe.com, but https://www.google.com or https://www.twitter.com or https://www.facebook.com will not work. If I hit F1 on the keyboard and bring up help in a new browser, I can connect to any site with no problems.

tailorvj, 2016-06-06:
I guess these sites are actively blocking such apps from loading them. 
I'm working on a new solution that will be more loosely coupled and based on a reactive node.js server and chrome extension that loads pages directly into the browser. 
If you are and/or you know JS developers who would like to participate in development I'd be more than happy. I have collected most of the research material for the solution.

rharkonen, 2016-06-06:
Ok, I just tried creating an iframe on my computer and pointing it towards the internal website and it would not load! Glad to know that I am not going crazy. I don't know too much JS so I wouldn't be of much help. I might have to try a different solution for what I need for now.

Thanks!

tailorvj, 2016-06-07:
@rharkonen Please open a feature request saying you would like a dashboard solution that isn't based on iframes. I'll take it from there and you will be updated once it is published (soon I hope)

Cheers,
Tailor

JanStevens, 2016-06-16:
Maybe use following extension? https://chrome.google.com/webstore/detail/revolver-tabs/dlknooajieciikpedpldejhhijacnbda?hl=en

anthonywebb, 2016-10-19:
I can confirm that iframes in chromium do not work :(  Anyone know if this is fixed in the most recent version of chromium (pixel?)  I have a server side nodejs app that I am sending commands to a locally hosted page telling it what the src of the iframe should be, allowing me to rotate the content programatically.  Is there any other way to do this?

This guy used iceweasel to get the job done. https://www.raspberrypi.org/forums/viewtopic.php?t=97356&p=678210

guysoft, 2016-10-20:
You could switch to iceweasel by installing it and changing the browser running command.

Edit the file ``/home/pi/scripts/start_chromium_browser``

The source for comparison is here:
https://github.com/guysoft/FullPageOS/blob/devel/src/filesystem/home/pi/scripts/start_chromium_browser

jounivarjonen, 2016-11-20:
Which steps are required for getting Weasel instead of chromium to autolauch? 

I installed Weasel but could not find the place where set weasel to startup instead of chromium.

guysoft, 2018-07-25:
This is out of date. The solution is here at the bottom, closing if no reply