guysoft/FullPageOS #32: Hiding scrollbars

bakteria1, 2016-06-07:
Is there any way to hide scrollbars globally?

tailorvj, 2016-06-07:
Please try to install the following Chrome extension to the chromium browser on FullPageOS and tell me if it satisfies your use case

Kind regards,
Tailor

bakteria1, 2016-06-08:
The chrome store would not let me download the extention since "chrome was out of date"

tailorvj, 2016-06-08:
We'll update chromium for the next release and see how it affects this issue. Could you please open this as a new issue decribing you inability to install extensions from the webstore due to the "chrome was out of date" issue?

guysoft, 2016-06-13:
Can you write down what version is needed and what is the current version, so we can confirm its in 0.6.0 RC1?

mrozendaal, 2017-03-03:
Im also looking for a solution to hide the scrollbar, default you could use "disable-overlay-scrollbar" but i dont know where to put this.

guysoft, 2017-03-05:
Overlay scrollbar means you have scrollbars that take up no active screen real-estate. A scroll appears when the pointer is in proximity to the scrollbar.
Enabling overlay scrollbar might actually hide it.
Here is how to do it:
http://www.skipser.com/p/2/p/enable-overlay-scrollbar-google-chrome.html

tdusmoha, 2017-10-27:
The other way is to hide the scrollbar using CSS. Something like this worked for me

<style type="text/css">
body {
    overflow:hidden;
}
</style>

G3z, 2018-02-15:
@tdusmoha even if you do this the greyed scrollbars appears.
@mrozendaal thank to you i was able to fix this.

just edit `/etc/chromium-browser/customizations/00-rpi-vars` and add `--enable-overlay-scrollbar` (or disable)

@guysoft : wouldn't it be nice to reference this file in the readme (if someone has a link to a list of chomium flags it would be nice too)

guysoft, 2018-02-16:
@G3z sure. PRs are welcome :)

G3z, 2018-02-19:
@guysoft I tought a wiki would be more fitting https://github.com/guysoft/FullPageOS/wiki/Chromium-configuration I hope you don't mind

guysoft, 2018-02-19:
@G3z excellent

IXIcOrDLeSsIXI, 2023-08-16:
Hello, first time posting on GitHub. I stumbled upon this as my company wanted to run a website with various marketing related images in our lobby and I was able to get it set up and work fine, however the scroll bars are definitely bugging me. I don't know what file to edit based on the above info. Nothing I see on the SD card matches. Can someone help with more detailed instructions?

guysoft, 2023-08-20:
@IXIcOrDLeSsIXI Hey and welcome to github.
The file is located on the raspberrypi on /etc/chromium-browser/customizations/00-rpi-vars . You will need basic Linux knowledge to edit that file from within the operating system on your Pi. I am not able to cover beyond that because its out of the scope of this repository.