guysoft/FullPageOS #33: How to serve a local file?

oherst, 2016-06-07:
I'm using version 0.5.0 on a Pi model B and trying to serve a file on the SD card but Chromium keeps giving me an error 'FILE_NOT_FOUND'. I might just be getting the URL format wrong, I'm not sure. In fullpageos.txt I've set the following:
file:///~/test/index.html

Any clues? Many thanks.

tailorvj, 2016-06-07:
Hi, there's a newer version and I suggest you use it. We are at very early stages of rolling this baby out to the world and we're doing it with very few resources, so I believe we will be able to currently support only the latest version. 
Beyond that, I think you should SSH into the the Pi and upload files to a folder under webroot, then reference them as http://localhost/folder/filename.html. 
webroot is located at `/var/www/html`

oherst, 2016-06-09:
Ok thanks, we decided to roll the newest version on a Pi3 anyway.

closed 2016-06-09 by oherst