guysoft/FullPageOS #36: How to set FullPageDashboard tabs

tailorvj, 2016-06-08:
Connect to  http://fullpageos.local/FullPageDashboard/ from another computer on the same network and edit dashboard tabs as you wish, then restart the raspberry pi

You can set the time for each tab by clicking the timer icon on the top left

capitanbatata, 2017-06-22:
What if you want to do this programatically? I see there is a [urls.json](https://github.com/amitdar/FullPageDashboard/blob/master/api.php) file that gets read, but no idea where this JSON file is supossed to be placed...

guysoft, 2017-06-22:
@capitanbatata YOu can edit urls.json, and it will only be considered in the first boot (So you need to edit it before you boot).
Also there is an api script in php.

capitanbatata, 2017-06-22:
Thanks. The question is in which directory this file has to be placed (`root?`, `/vat/www/`?). And what is the format? I assume something like

```
[
{
  "title": "Some title",
  "url": "some url"
},
...
]
```

guysoft, 2017-06-23:
I think from what you are describing you can just use ``/boot/fullpagedashboard.txt``

esaruoho, 2017-08-11:
I'm confused. Can I just run multiple URLs on multiple lines to `/boot/fullpagedashboard.txt` or am I stuck with the clunky index.php created for FullPageDashboard - which shows a "tab" and doesn't show the pages at full size (resized) - instead showing a scrollbar on the right and a tab with the timer on the top corner.

guysoft, 2017-08-11:
@esaruoho You can stop using FullPageDashbaord, or you can write better code for it.

jmcgeheeiv, 2018-06-21:
`urls.json` is in `/var/www/html/FullPageDashboard/urls.json`.   Modify this file to directly control the multiple URLs displayed.

I saw in one of the JavaScript files that each page is displayed for 30 seconds.  Search for "30" and replace this with the number of seconds you want.

thecuriouskid, 2018-08-08:
@guysoft May I ask if there is a way to have different URLs located at /boot/fullpageos.txt?

Also when adding another Tab on the FullPageDashboard, I am getting this error below. What could be the reason about it? Thanks. 
[image]

guysoft, 2018-08-08:
@thecuriouskid Please read the issue.
Regarding your url. please provide it.

thecuriouskid, 2018-08-08:
@guysoft Sorry for the confusion. What I mean is that could I insert multiple URLs in /boot/fullpageos.txt and where they will be switching from time to time? Like this one below. 
[image]

thecuriouskid, 2018-08-08:
@guysoft Yes, I understand that it should be under that one. But could it also be done at /boot/fullpageos.txt?

guysoft, 2018-08-08:
No, that would make the configuration files ambiguous.

jarsiv, 2019-07-23:
I tried to display multiple URLs with FullPageDashboard.
By editing /boot/fullpagedashboard.txt I got 404 - not found error. This for example http://www.google.com. With CTRL+T and entering the same address to it works. 

What might be the problem?

I also tried to modify /var/www/html/FullPageDashboard/urls.json, obviously it doesn't still work.

Displaying one external URL works fine with /boot/fullpageos.txt

guysoft, 2019-08-05:
@jarsiv Use teh gui. done edit ``/boot/fullpagedashboard.txt`` , that only works on first boot

varac, 2020-02-07:
@guysoft Can you please udpate the `README.md` with a clear documentation how to provision fullpageos with multiple URLs which gets cycled ? I think this is a very popular use case and even after reading this issue (after I found it) I'm still not sure how to do it properly (i.e. do the URLS need to get slashes escaped ?).

guysoft, 2020-02-07:
@varac Yes that should be done and would close this old issue. BTW pull request are welcome, its not high priority here.

DarrylHodges, 2023-07-21:
when you edit the fullpageos.txt and point it to the fullpagedashboard, make sure you type it in case correct, as it is case sentitive.
Then you can browse to the site and add in the sites with the gui.

guysoft, 2025-10-07:
Updated here:
https://github.com/guysoft/FullPageOS/wiki/FAQ#can-i-set-multiple-urls-in-bootfirmwarefullpagedashboardtxt-

closed 2025-10-07 by guysoft