guysoft/FullPageOS #37: Please share your public dashboards

tailorvj, 2016-06-09:
Dear FullPageOS users, we'd like to showcase (and use for tests) your public dashboards. If you are running any public IoT, monitoring or whatever monitor on FullPageOS, please share it with us in the comments below and we'll use it for our tests. 

Kind regards,
Tailor

mcfrojd, 2016-06-14:
You can find many dashboards on https://www.thedash.com/explore
Even make your own (1 free)

tailorvj, 2016-06-14:
Thanks @mcfrojd . I've just created a dashboard for FullPageOS Github repo :)
https://www.thedash.com/dashboard/BJ8GUOaN/fullscreen

tailorvj, 2016-06-17:
Chartbeat realtime analytics dashboard demo
https://chartbeat.com/demo/

peteruithoven, 2017-02-03:
https://github.com/peteruithoven/KickstarterStatus/

FloppyKing, 2017-02-28:
http://empyr.net23.net/TabbedPages/

simple tabbed pages.  Built for computers at work used to display page to clock in/out and the schedule.  Built in tabbed page functionality had ability for user to alter tabs, so didn't want to use that.  Also this page has a built in screensaver that kicks in after 1 minute since these are hooked to CRT monitors and didn't want them to be burned in.  Would be simple to add or alter tabs for more pages.

SecularSteve, 2024-02-07:
We have a Discussion Board now. In case you see this.