guysoft/FullPageOS #39: WIRED STATIC IP:  cannot connect via ssh

e-verdillon, 2016-06-09:
Hi,
My network conf:  fullpageos-network.txt
_#auto eth0:1
iface eth0:1 inet static
address 192.168.1.240
netmask 255.255.255.0
gateway 192.168.1.1_

Ican't connect to the PI via SSH, 
I use the img:
http://docstech.net/FullPageOS/nightly/2016-05-27_2016-05-10-fullpageos-jessie-lite-0.6.0.zip

Regard from French Alps,
Emmanuel

dhhdev, 2016-06-14:
Try and connect a keyboard/mouse to the Raspberry Pi. Press ALT+F# (anything but 1) and try to login.
Execute `ip addr` and see what information you get back to help diagnose the problem.

Also, if you use WiFi, the static method in `fullpageos-network.txt` won't work. You'll have to do something different for Wireless connections.

mcbvmware, 2016-07-07:
I get 'Connection Refused' on a wireless, DHCP setting; I'll login with a keyboard and find out what is going on.

mcbvmware, 2016-07-08:
OK, so sshd hadn't been started. Let me dig into it. Once I'd logged in and started sshd everything was fine.