guysoft/FullPageOS #40: Black screen with only cursor

DPH, 2016-06-14:
Great project.

I just can't make it start - using wired ethernet Pi, using everything as supplied (latest build)  just get black screen with cursor on it. Can SSH in so network working.

Not sure what else to try - any help welcome.
Cheers David

dhhdev, 2016-06-14:
Did you build FullPageOS yourself, or did you use the provided image by the mirror? Also providing some more info about your Raspberry Pi might be helpful I reckon. (Version of Raspberry Pi etc.)

DPH, 2016-06-14:
Thanks - used the mirror image2016-05-27..., and tried the nightly - which appears to be the same.
Using Raspberry Pi Model B using wired LAN.

Regards David

dhhdev, 2016-06-14:
That's weird. By using a wired connection, no setup should be needed. It should simply start the first time, expand the SD card, reboot - and then show you the welcome screen. (At least it did this for me, both on old Raspberry Pi 1 and 2.)

Did you change any files on the SD card before trying it out? (txt files, etc.)

tailorvj, 2016-06-14:
As far as I know, 0.60 RC1 runs on Raspberry Pi 2 and above because it uses Jessie, which is designed for ARM 7 CPU

DPH, 2016-06-14:
Not changed any files.
As it is an ARM6  (pi 1 model B) does that mean Jessie won't run properly?

Would it work with the Pi 3 (Arm 8 processor?)

dhhdev, 2016-06-14:
Well, I did build FullPageOS myself and could get it to run fine. (Build on a more or less clean install of the latest Raspian Jessie Lite.) I've Model B Revision 2.0. I'll test some more later today.

DPH, 2016-06-14:
Thanks, is it my Pi is too old (in which case I will get a Pi3) or is it the mirror image doesn't work but a build does?
Cheers David

tailorvj, 2016-06-14:
The mirror image is certainly built for ARM7 and there is no chance it would run on ARM6 processors as far as I know. 
Building the current version for ARM6 could possibly work. I guess @guysoft could provide further instructions and insights on that

dhhdev, 2016-06-14:
Yes and no. Depending on the hardware you build things, stuff might change. For example, think that if you build something for an Intel CPU, it would work in another way on an AMD. Stuff changes. I've just build a standard FullPageOS using my Raspberry Pi Model B Revision 2. (ARM6.)

If it works as intended, I wouldn't mind uploading it somewhere for you.

tailorvj, 2016-06-14:
@dhhdev That sounds great. Please keep us updated

DPH, 2016-06-14:
Would be great to share it if it works ok for you. 
Cheers David

mcfrojd, 2016-06-14:
I have had the same problem today, my pi 3 at home works fine, but my pi 2 a work becomes black with cursor.
Using the latest nightly.
When i go to the 192.168.1.xxx/FullPageDashboard/ from another computer, the default tab wount work, if i add google.com to a tab, it works, but trying to add a "thedash" https link dont work.
(google.com only works on the computer with 192.168.1.xxx/FullPageDashboard/ in the browser, not the screen with my pi.

dhhdev, 2016-06-14:
The chromium-browser seems to be at fault here, after debugging for a long time I came to that conclusion. Apparently newer versions of chromium-browser can't be build without SSE2 support, which ARM6 doesn't have.

Changing the browser to Midori got me a working FullPageOS on my Raspberry Pi Model B Revision 2.

guysoft, 2016-06-14:
Thanks Daniel for digging that info!

dhhdev, 2016-06-14:
@guysoft You're welcome. I'm actually going to make a Midori version (ARM6) myself in the upcoming days. And of course, sharing the code. Stay tuned.

DPH, 2016-06-14:
@dhhdev thanks Daniel. How can I change the browser?

Cheers David

dhhdev, 2016-06-14:
@DPH If you're not into Linux. This might be hard to explain. (I don't know how your level is.)

But to put it simply, I did three things:

1.) I uninstalled the Chromium with the `apt-get purge`-command.
```
bash
pi@fullpageos:~ $ sudo apt-get purge chromium-browser
```

2.) I installed the Midori browser.
```bash
pi@fullpageos:~ $ sudo apt-get purge midori
```

3.) I changed the `~/scripts/run_onepageos` file.
```bash
pi@fullpageos:~ $ vim ~/scripts/run_onepageos
```

```bash
#!/bin/bash
xset s off         # don't activate screensaver
xset -dpms         # disable DPMS (Energy Star) features.
xset s noblank     # don't blank the video device

matchbox-window-manager &
x11vnc -many -rfbauth /home/pi/.vnc/passwd &
while true
do
    if [ $(curl -sL -w "%{http_code}\\n" "http://localhost/FullPageDashboard" -o /dev/null) == "200" ]; then
      (sleep 15 ; /home/pi/scripts/fullscreen) &
      xdotool mousemove 9000 9000
      midori -e Fullscreen -a $(head -n 1 /boot/fullpageos.txt)
    fi
    sleep 1
done
```

This is not cleaned up by far... there are many more things I could have done. But to test if it even would work, if it was just the browser. That's the steps I took.

I plan to make a Midori browser version of FullPageOS for people running ARM6. (I run ARM7 at work. But at home I use my old Raspberry Pi.) So why not give back to an awesome project. 👍  I hope this helped, until I get a more polished solution worked out.

guysoft, 2016-06-14:
It might be worth writing this down as a variant.
I can add a variable that would make chromium install optional. And then
you can set it not to install it in your variant and install midori.
Modified scripts could also be shipped in the variant.

dhhdev, 2016-06-15:
@guysoft That was the plan. To simply make it a variant. I'll have a crack at it later today, right after I get some sleep. 👍

mcfrojd, 2016-06-16:
I have tried this Midori version on my Pi2 at work.
Now i can get some pages to show, but it wount show a https:// url that works on my pi3 at home

mcfrojd, 2016-06-16:
With this "midori" switch i was able to get my https://thedash adress to work if started from fullpageos.txt
The fullpagedashboard seems to have a problem showing https:// links thou.

dhhdev, 2016-06-16:
@mcfrojd Regarding titlebars and stuff with Midori, it should hide. There's flags for this. Actually better than Chromium. Where you have to switch with a emulated key press in the code. (F11.)

`midori -e Fullscreen -a https://my.awesome.webpage.com`

mcfrojd, 2016-06-16:
Well i run midori as your example midori -e Fullscreen -a $(head -n 1 /boot/fullpageos.txt)
But i still get a small titlebar.
https://www.dropbox.com/s/5m5y51rsmp3j5vl/Sk%C3%A4rmklipp%20Grantek%202016-06-16%2012.38.52.png?dl=0

mcfrojd, 2016-06-16:
when i run "midori –help-execute" in putty i only get a "Midori - Cannot open display:" message so i can't list all flags

guysoft, 2016-06-16:
Run
DISPLAY=:0 midori -help-execute

It would redirect midori to the X server if there is a graphical interface

dhhdev, 2016-06-16:
@mcfrojd I might have changed some other files too, and just simply forgot it - since I was a bit busy with two images at the same time and some work related stuff. I apologize. But remember, this is only a temporary solution to get things working on Raspberry Pi's before version 2. (The ones running with ARM6 processors.)

Inside the `~/.scripts/fullscreen` and `~/.scripts/refresh` files, try and change this line:

`WID=$(xdotool search --onlyvisible --class chromium|head -1)``

To this:

`WID=$(xdotool search --onlyvisible --class midori|head -1)`

Or simply run this to change the files, with the pi user:

`sed -i.bak s/chromium/midori/g ~/.scripts/fullscreen && sed -i.bak s/chromium/midori/g ~/.scripts/refresh`

mcfrojd, 2016-06-17:
@dhhdev I did the changes (the manual way.) But the small titlebar (or what it is, is still there).

Tried to comment out "matchbox-window-manager &" in the run_onepageos file (was not sure if i had added that one my self or not. Doing that removed the "titlebar" but the screen is no longer fullscreen then.
https://www.dropbox.com/s/dx9r99v6rbixnp9/fullpageos_not%20fullscreen.png?dl=0

dhhdev, 2016-06-17:
@mcfrojd That's odd. It should work. Actually even without the two files there. I'm working on a variant for older Raspberry Pi's. (ARM6.) At the moment. Which will be using the Midori browser. So I can clean it up. For now, there is one thing you can test.

In your VNC viewer, try and hit F11. If it works, I know what the culprit is.

Look in `~/scripts/fullscreen` and `~/scripts/refresh` again.

Remove this line:
`xdotool key F11`

dhhdev, 2016-06-17:
About the `https://thedash` error. Since it's not a real domain, and I suspect an website on another server on your local network. You'll need to add the following line to the `/etc/hosts`file:

`x.x.x.x	thedash`

The IP part (x.x.x.x) should be changed, if it's running on another computer/server on your local network to the local IP address of that network.

You can find your local ip address by executing the `ip addr | grep inet`-command on the server that is running the webpage you are trying to display. For example, the output could be:

...
**inet 192.168.87.50/24 brd 192.168.87.255 scope global wlan0**
...

^ This is my local network IP address of my Wireless Conection on my Raspberry Pi.

mcfrojd, 2016-06-17:
@dhhdev It was the F11 that made the titlebar show.

About the "thedash" link its offcource a real link like @tailorvj :s link https://www.thedash.com/dashboard/BJ8GUOaN/fullscreen
I just did not want to share the link to my workplace dashboard.
The real "thedash" link works fine if i enter it in fullpageos.txt, but not if i add it to the fullpagedashboard (Going to the local ip/fullpagedashboard from another pc and adding it to a new tab)
Maybe its the https (and not http) thats making the problem there?

dhhdev, 2016-06-17:
@mcfrojd Argh, I see. I wasn't familiar with the **thedash** service. Well, there could be a few different reasons for why it won't show properly. And to my knowledge, FullPageDashboard is going under heavy development. To get something new and better, but don't quote me just yet.

I'll test some with the **thedash** service. To see if I can make some sort of solution, while I'm at the other stuff. At least something temporarily.

If you would make another issue for that. (**thedash** not working.) It would be easier to keep track of this issue alone. It might also affect other services/sites.

DPH, 2016-06-20:
Thank you, works great.

sweenig, 2016-07-30:
I've tried both this image (http://docstech.net/FullPageOS/2016-07-20_2016-05-27-fullpageos-jessie-lite-0.7.0.zip) and this image (http://docstech.net/FullPageOS/nightly/2016-05-27_2016-05-10-fullpageos-jessie-lite-0.6.0.zip) but i can't seem to get either to work with my Raspberry Pi Model B+. I assume it's because both these images are built for ARMv7 and the Raspberry Pi Model B+ runs ARMv6.  Has anyone built an image? Or can some one write simpler instructions (for those of us who do not compile from source very often) so that it can work on ARMv6? Am I just using the wrong image?

guysoft, 2016-07-31:
@sweenig Pi1 B+ is unsupported, please see the feature list,  #12 and #43

guysoft, 2017-08-20:
Closing if anyone doesn't have anything to add

guysoft, 2017-10-23:
Fixed, it works now, its slow as hell and takes 5-10 minutes to boot. Might work as a digital picture frame.

closed 2017-10-23 by guysoft