guysoft/FullPageOS #41: raspbian-jessie-lite: Missing lsof package

dhhdev, 2016-06-14:
Just a little heads up, if you want to build FullPageOS yourself. You'll be missing `lsof`. At least if you run the latest version of the Jessie Lite version of Raspbian. (As of writing this.)

A simple `sudo apt-get install lsof` will fix this.

I would suggest adding this to the README.rst. At least the `chroot_script` will be happy then. 👍

guysoft, 2016-06-16:
Will use that image for 0.7.0, will add it then.

dhhdev, 2016-06-16:
Awesome, good to hear. A small thing I noticed, that was missing when building from a Jessie Lite image. 😄

guysoft, 2016-06-17:
Why did you close?
A bug should be closed once it is fixed. Not when its decided to be fixed.

dhhdev, 2016-06-17:
Argh. Sorry. My brain wasn't working at the time, so I didn't think about it. I had been up since midnight, and crashed when I came home from work. Apologies.