guysoft/FullPageOS #42: unclutter instead of xdotools

dhhdev, 2016-06-14:
Looking through the code, I noticed FullPageOS using xdotools to move the mouse off the screen. I would just argue that unclutter would be a bit more clean - and probably more lightweight. Is xdotools used for anything else?

FullPageOS/src/filesystem/home/pi/scripts/run_onepageos

```
#!/bin/bash
xset s off         # don't activate screensaver
xset -dpms         # disable DPMS (Energy Star) features.
xset s noblank     # don't blank the video device

matchbox-window-manager &
unclutter &
x11vnc -many -rfbauth /home/pi/.vnc/passwd &
while true
do
    if [ $(curl -sL -w "%{http_code}\\n" "http://localhost/FullPageDashboard" -o /dev/null) == "200" ]; then
      (sleep 15 ; /home/pi/scripts/fullscreen) &
      chromium-browser --kiosk --app=$(head -n 1 /boot/fullpageos.txt)
    fi
    sleep 1
done
```

tailorvj, 2016-06-14:
As you can see in previous threads regarding the mouse arrow issue, we intend to use unclutter in future releases. Please feel free to implement and send a pull request

Kind regards,
Tailor

dhhdev, 2016-06-14:
Awesome. I just wasn't sure if xdotools wasn't used for any other automatization.
I'm actually just making a dev environment on my old Raspberry Pi. (The first one. Jessie Lite Raspbian.)

I'll of course share whatever I might fix in the process of my own project. :-)

closed 2016-06-14 by dhhdev