guysoft/FullPageOS #44: FullPageOS 0.6.0 RC2

guysoft, 2016-06-15:
Fixes RC1 issues:
Vnc password issue #26 
Mouse movement #34 

Image:
http://docstech.net/FullPageOS/2016-06-15_2016-05-10-fullpageos-jessie-lite-0.6.0.zip
Md5:
http://docstech.net/FullPageOS/2016-06-15_2016-05-10-fullpageos-jessie-lite-0.6.0.zip.md5

Please confirm there are no new issues and we can finally release it

DPH, 2016-06-15:
Thanks - will try it tommorow.

guysoft, 2016-06-18:
@DPH @tailorvj any news? Its kinda hard to know if this works without QA. I tested it here, but I might have missed something.

tailorvj, 2016-06-20:
@guysoft I just burned the image and it works on my Pi 2. 

BUG: The small presentation bug on FullPageDashboard is back. I don't really mind because I'm going to replace it, but I guess there was a previous solution to this problem and you know what it was

tailorvj, 2016-06-20:
BUG: /boot seems to be write-protected. I can't save changes to .txt configuration files

guysoft, 2016-06-20:
Not sure if that bug was also present in RC1

tailorvj, 2016-06-20:
AFAIK it wasn't present in RC1. When I booted to RC1, the welcome presentation displayed well

guysoft, 2016-06-20:
The scale bug is random. It would be nice to get more reports

tailorvj, 2016-06-21:
I'm not too concerned about it because the dashboard is going to be replaced in 0.7.0 and there will be no iframes involved

guysoft, 2016-06-21:
Sounds like we should release and get going to 0.7.0. How far are you with that dashboard?

felipog, 2016-06-22:
Hi!
I just want to properly thank you for this distribution. It works perfectly for what I need. I have 4 Pi 2 running the official releases, and now two more Pi 3 running this latest build.

So thank you!

tailorvj, 2016-06-22:
@guysoft I'm prepared with the architecture and research materials. I've recruited a couple of developers for the task in the past few days. We will run a hackathon in upcoming weeks and then test it out for 0.7.0

guysoft, 2016-06-22:
ok, if there are no blockers im releasing this and preparing for 0.7.0

guysoft, 2016-06-30:
Released, download link is no longer available since its now the release:
https://github.com/guysoft/FullPageOS/releases/tag/0.6.0

closed 2016-06-30 by guysoft

matt7277, 2016-08-10:
I am using the 0.6.0 release and it looks like I randomly started getting the scale bug with a RaspberryPi 3 + the official raspberry Pi touchscreen display. 

What is strange is that things worked perfectly fine for the past 10+ days. 

I added NASA APOD to the rotating dashboards and all of a sudden the scale blew up and refuses to go back to normal :( any idea on how to fix the scale bug?

guysoft, 2016-08-14:
@matt7277 Then open a bug, this one deals with 0.0.6 RC2.