guysoft/FullPageOS #46: Not quite FullPage Dashboards

lithodora, 2016-07-06:
I loaded up the the most recent Jesse version onto my Rpi 2.

In order to get the HDMI to work I had to change the config to uncomment:
`hdmi_force_hotplug=1`

The result can be seen here:
[Image can be viewed here](http://i.imgur.com/Nt6iQaI.jpg)

I have used different monitors with fresh install on a new SD card. Same results.

The mouse can move over the full range of the screen. It IS fullscreen. The dashboards appear to be smashed into 10% of the screen.

Any ideas?

guysoft, 2016-07-06:
Iv'e seen this bug. Its in FullPageDashboard iframe bug. A workaround would
be to disable it.
It happens at random, so I had little luck isolating it. A restart tends to
fix it too.

lithodora, 2016-07-25:
I have tried to use the newer build. Same result.  A restart never solves it.

> Its in FullPageDashboard iframe bug. A workaround would be to disable it.

 How do I do that?

guysoft, 2016-07-25:
1. What version of raspberrypi are you using?
2. You can change the url that is loaded by changing ``/boot/fullpageos.txt``

lithodora, 2016-07-25:
I have a rPi2 Model B v1.1. 

I can't repeat it for you to fix it, but it sorted out in the end...
I wrote the image to 5 different times until one of them worked fine. It has worked consistently every time.
I read from that to create an img that I am able to use. No idea. I only changed that one setting:
`hdmi_force_hotplug=1`
and of course the network.

closed 2016-07-25 by lithodora

guysoft, 2016-07-26:
Strange, rewriting the image is not supposed to effect that. Might be a cashing issue, but no way to tell.

lithodora, 2016-07-27:
The SD cards I have are cheap. Rebooting the pi can cause the card to be unreadable. I put the card in windows it says it needs to fix it and does. Put it back in the pi and boots up... Let's blame the cards.