guysoft/FullPageOS #48: Use Synergy with FullPageOS

mcfrojd, 2016-07-11:
I have been following the guide at http://www.etcwiki.org/wiki/Install_Synergy_Raspberry_Pi_3_-_June_2016 to install Synergy on my pi with FullPageOS.
But i seem to get stuck on how to get the "autostart file" ~/.config/autostart/synergy.desktop
to autostart?

If i type cynergyc 192.198.1.22 manually in putty the Pi connects to my synergy server, but how do i get it to autostart. I guess FullPageOS bypasses the ability to have that kind of autostart file, but i guess there is another way to autostart the line /usr/local/bin/synergyc 192.168.1.22

tailorvj, 2016-07-17:
@guysoft should be able to answer that, although it's out of scope for FullPageOS, but I think it's interesting :)

guysoft, 2016-07-18:
1. It might work putting the .desktop in ``~/.config/autostart``. I haven't tested
2. Another way might be to add the line to ``/home/pi/scripts/run_onepageos``
Add it after ``x11vnc -many -rfbauth /home/pi/.vnc/passwd &`` and make sure to add a ``&`` at the end so it won't get stuck
3. We might want to add a way to add .desktop files on FullPageOS

guysoft, 2018-07-25:
Closing if no reply