guysoft/FullPageOS #49: New Tab Stuck Full Screen

Raeki, 2016-07-12:
When I open a new tab in FullPageOS and make it full screen by pressing F11 pressing F11 again to exit fullscreen does nothing.

tailorvj, 2016-07-17:
Set the URL you want in the file fullpageos.txt on the SD card and reboot the Pi

guysoft, 2016-07-18:
Can anyone confirm this?

@tailorvj He is describing a bug we need to confirm.

@Raeki Please confirm these are your steps:
1. open new tab using ctrl-t
2. open a web page in that tab
3. press F11 to make that tab full screen
4. press F11 again to make that tab not full screen

expected: the new tab should not be at full screen
actual: new tab remains at full screen

tailorvj, 2016-07-19:
@guysoft I wouldn't say this is a bug in FullPageOS, since this isn't something the system is expected to do. The whold concept is not to connect a keyboard or mouse to the system at all. It should be running 1 URL which is set in fullpage.txt on the SD card. 
We are going to have URL rotation functionality in 0.7.0 using a Chrome extension and a Node.js app running on the Pi. Expect this beginning of August

Raeki, 2016-07-19:
@guysoft 
Sorry for my delayed response, your description of the issue is correct. Thank you. 

@tailorvj 
I understand how the OS is generally meant to be used. Unfortunately some of the websites I want to be displayed don't play well with iframe so I need to open a new tab instead. Although at this point I may as well end up using regular old Raspbian since I've just ended up using a few chrome extensions to gain the same functionality.

guysoft, 2017-09-26:
Closing if no reply