guysoft/FullPageOS #51: Removing lighthttpd

cjus, 2016-07-15:
I'm running a NodeJS app on my Pi and don't have a need for lighthttpd/PHP. Is there a way to remove lighthttpd without building a FullPageOS variant?

guysoft, 2016-07-17:
``sudo apt-get remove --purge lighttpd`` ?

tailorvj, 2016-07-17:
@cjus Would you like to help us integrate Node.js + Express http server instead of lighttpd for 0.7.0? We are having a hackathon on July 29-30th

guysoft, 2016-07-18:
I can make lighttpd optional with a config setting, which would let you build a nodejs variant with ease.

tailorvj, 2016-07-19:
@guysoft sounds great. Do you think we can have the node.js code running from the SD card? That way we could replace the node app code easily without touching the system

commit 3d567406 (2016-07-20): Add LIGHTTPD, FULLPAGEDASHBOARD and WELCOME switches to config, fixes #51
closed 2016-07-20 by guysoft