guysoft/FullPageOS #53: FullPageDashboard v2.0 hackathon on July 29-30th

tailorvj, 2016-07-17:
Join us (virtually or physically) for the next phase of FullPageDashboard development

We are looking for developers in the following fields of knowledge:

- Node+Express websocket
- Chorme extensions
- Front-end mobile web
- System Linux (shell scripting)

Hackathon details: [http://hacktheweb.co/](http://hacktheweb.co/)
Start: July 29th, 11am GMT +02:00
End: July 30th, 6pm GMT +02:00

Contact [tailor.vj@gmail.com](mailto:tailor.vj@gmail.com) or [http://m.me/fullpageos](http://m.me/fullpageos)

guysoft, 2016-08-01:
@tailorvj do you want to close since after our talk?

guysoft, 2016-09-08:
@tailorvj are you planning to do this in another time?