guysoft/FullPageOS #57: Build error during image resize

semanticfire, 2016-07-21:
While using the build script I run into this error ( end of log )

Initial idea, language issue ??

=========
Unmounting /opt/FullPageOS/FullPageOS/src/workspace/mount...
Resizing partition 2 on 2016-05-27-raspbian-jessie.img to minimal size + 200 MB
e2fsck 1.42.9 (4-Feb-2014)
Stap 1: Controle van inodes, blokken, en groottes
Stap 2: Controle van mappenstructuur
Stap 3: Controle van verbindingen tussen mappen
Stap 4: Controle van verwijzingsaantallen
Stap 5: Controle van groepssamenvattingen
/dev/loop0: 92508/305824 bestanden (0.1% niet-aaneengesloten), 523896/1220096 blokken
/opt/FullPageOS/FullPageOS/src/common.sh: regel 309: * 4096: syntaxfout: operator verwacht (het onjuiste symbool is "* 4096")

guysoft, 2016-07-21:
I suggest you build using the vagrant method as listed in the readme. It
solves envoriment issues like this. I created it after having a similar
issue.

semanticfire, 2016-07-21:
Okay I'll try to beat vagrant into submission, this throws tons of errors at me right now, probably some version issue

semanticfire, 2016-07-22:
I keep running into:

=====
==> default: /FullPageOS/src/workspace /FullPageOS/src
==> default: rm: 
==> default: cannot remove ‘2016-05-27-raspbian-jessie.img’
==> default: : Permission denied
==> default: 
==> default: exit 0 due to eval "$SCRIPT" 
==> default: BUILD FAILED!
==> default: unmounting image...
The SSH command responded with a non-zero exit status. Vagrant
assumes that this means the command failed. The output for this command
should be in the log above. Please read the output to determine what
went wrong.

guysoft, 2016-07-22:
1. Run vagrant as root as eexplained in the documentation
2. Delete the workspace, the old img there has root permissions and as the
error suggests it can't overwrite.

semanticfire, 2016-07-22:
Vagrant now works as it should and produces a workable image on my system.
Could you document how to use the variant mechanism with vagrant ?

guysoft, 2016-07-23:
Added to the readme, also pasting here:

To build a variant on the machine simply run:

```
cd FullPageOS/src/vagrant
run_vagrant_build.sh [Variant]
```

closed 2016-07-23 by guysoft