guysoft/FullPageOS #66: Jpgs not being displayed

lithodora, 2016-07-28:
Raspberry Pi 2 Model B v1.1

On the first loaded FullPageOS I noticed that some images were not being displayed in the demos. 
Some appeared and others were blank spots. I attributed it to lack of connectivity or some issue.

When I got my pages loaded none of the images loaded. After a little investigating:

[PNGs are fine.](https://cloud.githubusercontent.com/assets/20325156/17224977/ff0dfafe-54b7-11e6-8b15-b643340122d1.png)

 [JPGs not so much.](https://cloud.githubusercontent.com/assets/20325156/17224922/ca5b6a3a-54b7-11e6-9cf6-3bc9a9b87b41.png)

guysoft, 2016-07-31:
1. What power supply are you using
2. Sounds like the JPG library is corrupted - is there a chance you can reflash?
3. What version of FullPageOS?

lithodora, 2016-07-31:
1. I was using the wrong adapter for power: 5V .7A. In my defense that lettering is tiny and I thought it was a 2... I have swapped it for the correct power: 5V 2A. No change.

2. New card. Fresh flash. No change.

3. 2016-07-20_2016-05-27-fullpageos-jessie-lite-0.7.0
(I thought I'd give it a go... 2016-06-15_2016-05-10-fullpageos-jessie-lite-0.6.0 works perfectly)

guysoft, 2016-08-01:
1. More amps is better than less. That would not effect it
2. ok
3. Are you saying this only happends on 0.7.0 Nightly build? And that 0.6 works without this JEPG issue?

lithodora, 2016-08-01:
That is correct. I used 0.6 for a day or two before the update and switched. I never realized that the background images were not displaying on 0.7 until I started to try to use my own.

b1u3, 2016-08-04:
I got the same issue. Seems like a bug in this version of Chromium. To fix this, try a downgrade by removing chromium & chromium-l10n and install Chromium 50 manually: https://ubuntu-mate.community/t/tutorial-install-working-chromium-50/6762/1

guysoft, 2016-08-04:
If anyone can paste the version of chromium you are using which has the bug that could help.
Because we are using the ubuntu arm7 builds.

jasonbronson, 2016-08-12:
Seeing it in 51.0.2704.106 as well.

bkervaski, 2016-08-16:
Seeing it on Pi 3 using 2016-05-27-fullpageos-jessie-lite-0.7.0

uda, 2016-09-08:
Using @b1u3 's suggestion fixed the issue for me

yexingok, 2016-09-23:
We are on a brand new pi 3, and  we have the same issue: jpg not displayed

We use the latest stable version http://docstech.net/FullPageOS/2016-07-20_2016-05-27-fullpageos-jessie-lite-0.7.0.zip  which come with the Chromium version is 51.0.2704.106.

We are going to try the following:

A. downgrade Chromium from 51 to 50 refer here:https://ubuntu-mate.community/t/tutorial-install-working-chromium-50/6762/1
B. flush the previous stable (0.6.0) version:  http://docstech.net/FullPageOS/

yexingok, 2016-09-27:
Forgot to mention, solution A works fine for us, thanks everyone!

guysoft, 2016-09-28:
BTW, the reason I am not implementing this right away is that downgrading basically means that chromium would not update later on. 
Does anyone know if this issue is planned to be fixed on chromium's side?

guysoft, 2016-09-30:
Ok good news. Looks like the Pi foundation released a new build of chromium that they are bundling with the official raspberrypi.
[See release note:s for 2016-09-23](http://downloads.raspberrypi.org/raspbian/release_notes.txt).
``* Chromium web browser included``.
It seems like it will also use the hardware acceleration in the Pi for video playback. So this fixes a lot of issues at once

NoviCMS, 2016-10-27:
tried installing 53.0.2785.143-0ubuntu0.14.04.1.1142

the browser works... but the jpg's still not working.

guysoft, 2017-09-26:
Fixed in current versions.

closed 2017-09-26 by guysoft

s1h4d0w, 2019-05-01:
This issue has popped up once again on the most recent stable version. I have two jpg's set as background images and they didn't show. Swapping them out for PNGs worked.

guysoft, 2019-05-01:
@s1h4d0w Please provide the exact version you are using.

s1h4d0w, 2019-05-01:
@guysoft As far as I know the latest stable version, 2018-04-22_2018-04-18-fullpageos-stretch-lite-0.9.0, downloaded from http://unofficialpi.org/Distros/FullPageOS/

guysoft, 2019-05-01:
@ s1h4d0w Can you try using 0.10.0 RC3? #273

s1h4d0w, 2019-05-01:
@guysoft Will do as soon as I can, currently don't have access to an RPi to test