guysoft/FullPageOS #69: Wifi problems connecting

jasonbronson, 2016-08-11:
Keep trying different ways to connect to the wifi but it fails. I'm sure I have the right settings

cat /boot/fullpageos-network.txt 

```

# You can use this file to manually set up your network configuration.
#
# This file is included into /etc/network/interfaces, so anything that
# works by editing that file is also possible here.

### WIFI CONFIGURATION ######################################################
# The three segments below should cover you in most cases if you run
# a wifi network that uses either WPA/WPA2 or WEP encryption.
#
# Just uncomment the lines prefixed with a single # of the configuration
# that matches your wifi setup and fill in SSID and passphrase.

## WPA/WPA2 secured
iface wlan0-fullpageos net manual
    wpa-ssid “myssid”
    wpa-psk “my password

## WEP secured
#iface wlan0-fullpageos net manual
#    wireless-essid "put SSID here"
#    wireless-key "put password here"

## Open/unsecured
#iface wlan0-fullpageos net manual
#    wireless-essid "put SSID here"
#    wireless-mode managed

### WIRED CONFIGURATION #####################################################
# The following segment allows you to configure your wired connection
# with a static IP.
#
# Just uncomment the lines prefixed with a single #. Then connect 
# a cable to the Pi and another system, e.g. a Laptop, and set that
# other system's network configuration to:
#
#   address:   192.168.250.10
#   netmask:   255.255.255.0
#   broadcast: 192.168.250.255
#
# You can then reach the Pi from the system's browser by going to
#
#   http://192.168.250.1
#
# or
#
#   http://fullpageos.local

auto ethic:
iface eth0:1 net static
  address 192.168.1.80
  net mask 255.255.255.0
  broadcast 192.168.1.255
```
 

cat /etc/wpa_supplicant/wpa_supplicant.conf
```
country=GB
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
network={
    ssid="myssid"
    psk="my password"
}
```

ifconfig 
```
wlan0     Link encap:Ethernet  HWaddr b8:27:eb:8c:6a:28  
          inet6 addr: fe80::ff39:2fe5:edc7:ab32/64 Scope:Link
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:93 errors:0 dropped:93 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:39616 (38.6 KiB)  TX bytes:0 (0.0 B)
```

dmesg
```
[    6.575648] brcmfmac: brcmf_add_if: ERROR: netdev:wlan0 already exists
[    6.581212] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
```

guysoft, 2016-08-11:
Duplicate of #222  Your quotations are mac-style

jasonbronson, 2016-08-11:
I was able to fix the issue by changing this file
/boot/fullpageos-network.txt

And this line to remove the following from it "-fullpageos"

## WPA/WPA2 secured
iface wlan0 inet manual

Now I get an ip
`
wlan0     Link encap:Ethernet  HWaddr b8:27:eb:8*****  
          inet addr:192.168.1.166  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::ff39:2fe5:edc7:ab32/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:671 errors:0 dropped:260 overruns:0 frame:0
          TX packets:39 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:83187 (81.2 KiB)  TX bytes:7132 (6.9 KiB)`

closed 2016-08-11 by jasonbronson