guysoft/FullPageOS #72: Limited Browser size

tannert2, 2016-09-08:
I am trying to set up a dashboard to have multiple tabs rotating, but for some reason the page only takes up a small portion of the screen. Any fix for this? Below is an image of what I am seeing. 

[image]

guysoft, 2016-09-08:
Known bug. At the moment the only way around it is to not use FullPageDashabord. 
This can be done by chanigng the url in ``/boot/fullpageos.txt`` to your desired url. This can also be done on a windows machine.

tannert2, 2016-09-08:
Ok, thank you for your response. That is how I set it up the first time then thought I would try out the dashboard.  If you don't mind could you add an update to this issue thread when the problem is fixed.

guysoft, 2016-09-09:
Of course, will also link duplicates reports here. Its how we roll.
Also @tailorvj is sort if assigned to this. But we are all busy and it
would require replacing FullPageDashboard

guysoft, 2020-03-01:
This issue has been fixed with commit https://github.com/amitdar/FullPageDashboard/commit/59f847f447f3bac0369a4ff7fde6455b7e6cd1c5

closed 2020-03-01 by guysoft