guysoft/FullPageOS #74: Does it work with the new Raspberry pi zero?

jwillmer, 2016-09-28:


guysoft, 2016-09-29:
Duplicate of #12 and relevant to #43

closed 2016-09-29 by guysoft

jwillmer, 2016-09-29:
Maybe you should put it into the readme.

guysoft, 2016-09-29:
It IS in the readme, its the only bullet in requirements:
https://github.com/guysoft/FullPageOS#requirements

jwillmer, 2016-09-29:
Yes but it is missleading. There is no information about the ARM version and the Raspberry Pi Zero is not old it is the newest product.

guysoft, 2016-09-29:
You are more than welcome to add it to the README and pull request with text which seems better to you. I didn't want to include a lot of technical information such as the CPU architecture of unsupported devices.

remi-grumeau, 2016-09-29:
+1

This is a open source project.
It means you can participate and improve things if you think it's necessary.
Github makes that super-easy, just edit de readme file from their online editor, and hit submit.

Not just "give it to me for free as perfect as possible. now"

Remi

jwillmer, 2016-09-29:
Well guys come down, it is not like I ask for a new feature. I only suggest that it would make your life easier to put it into the README because I saw I am not the only one who created an issue because of this.

guysoft, 2016-09-29:
1. Those issues were created before rhe readme entry
2. Sometimes I don't know what to write, I get code or documentation
contributions for things I can't think of. In this case I don't know what
to write that would be considered "not misleading" to your eyes. This is
why I ask you to tell me what to write.