guysoft/FullPageOS #75: Use chromium bumdled with New Raspbian version 2016-09-23

guysoft, 2016-09-30:
Ok good news. Looks like the Pi foundation released a new build of chromium that they are bundling with the official RaspberryPI.
[See release note:s for 2016-09-23](http://downloads.raspberrypi.org/raspbian/release_notes.txt).
``* Chromium web browser included``.
It seems like it will also use the hardware acceleration in the Pi for video playback. So this fixes a lot of issues at once
This has a potential to fix the following issues:
#66 #29 #22

rojepp, 2016-09-30:
Looking forward to this. Wish I knew how to help.. :-/

guysoft, 2016-09-30:
At the moment the build is getting ``Hash Sum mismatch`` errors, so we need to wait till the repos start working again.

tailorvj, 2016-10-04:
It seems like we'd forecast the PIXEL release :) It also includes VNC and node.js out of the box

anthonywebb, 2016-10-11:
Has this been integrated yet in the nightly builds?  We are trying to set up some devices but would prefer to have the latest and greatest

anthonywebb, 2016-10-13:
@guysoft you think fullpageos will work with this latest release from pi foundation?  I'd like to give it a try when it is ready.  Thanks!

anthonywebb, 2016-10-19:
For those of you hoping that the new chromium build would solve the mp4 playback issues, it appears to have helped a bit, but its in no way a replacement for omxplayer, the new pixel interface is great though

guysoft, 2016-10-20:
Ok,
So here is a new build, I didn't pay with the chromium installation yet, so I encourage you guys to try and install the PIXEL installation which should be in the repos:
image: http://docstech.net/FullPageOS/nightly/2016-10-20_2016-09-23-fullpageos-jessie-lite-0.7.0.zip
md5 http://docstech.net/FullPageOS/nightly/2016-10-20_2016-09-23-fullpageos-jessie-lite-0.7.0.zip.md5

guysoft, 2016-10-23:
Ok, had a chance to test that nightly, there dosn't seem to be any`` chromium-browser`` package, only ``rpi-chromium-mods`` which has a flashplugin in it. Any insight would he helpful

NoviCMS, 2016-10-26:
Great, going to try the PIXEL installation!

guysoft, 2016-10-26:
I made another PIXEL installation one with the full image, that could also be tested, since I can't see a package for the PIXEL chromium-browser available:
http://docstech.net/FullPageOS/nightly/2016-10-23_2016-09-23-fullpageos-jessie-0.7.0.zip
http://docstech.net/FullPageOS/nightly/2016-10-23_2016-09-23-fullpageos-jessie-0.7.0.zip.md5

NoviCMS, 2016-10-26:
Oké, i a'm missing the hardware acceleration on my RPI 3, is there any way to enable this? are they working on it?

guysoft, 2016-10-26:
@NoviCMS 
1. What image are you using out of the 2?
2. How are you testing for acceleration? Are you talking about 2D or 3D?

NoviCMS, 2016-10-26:
@guysoft 
1: i am using the first image and the master branch
2: i am using the image for my RPI kiosk with touchscreen. (24" 1080p with toch overlay) 
This is my test site: http://in2shop.staging.novisites.nl/shop/526842
when opening a menu or scrolling down the page it becomes laggy. (i have tested many distro's on RPI3)

NoviCMS, 2016-10-27:
going to try 2016-10-23_2016-09-23-fullpageos-jessie-0.7.0.zip now. i hope the jpg bug is fixed.

NoviCMS, 2016-10-27:
This build: 2016-10-23_2016-09-23-fullpageos-jessie-0.7.0.zip is by far the best i have seen till now. maybe you could create a build from this one? if i boot the image the browser does not start. also fullpageos.local is not set.

guysoft, 2016-10-27:
@NoviCMS 
1. Can you be specific to what is good there?
2. The browser does not start in 2016-10-23_2016-09-23-fullpageos-jessie-0.7.0.zip? if so why is it the best?

NoviCMS, 2016-10-27:
Hey @guysoft 

1: video's play smooth(er), jps are displayed, overall experiënce is better.
2: The browser does not start automatically after boot, i was able to start it using the menu.

NoviCMS, 2016-10-27:
@guysoft can you create a img from the PIXEL installation with the full image with working fullpageos.local, and browser start?

NoviCMS, 2016-10-27:
@guysoft i see that the build has gone wrong, at the base there is a folder "filesystem" in there are the config files etc.... i will try to cp all to /

guysoft, 2016-10-27:
If there is a menu it means the special lightdm session isn't starting.
Thats the problem.

NoviCMS, 2016-10-27:
no success so far, i will look into it in a couple of days

matt7277, 2016-11-13:
Using the 10/23 Nightly build on a RaspberryPi 3 and also noticing that Chromium fails to launch. If you launch manually, the http://localhost/Welcome and http://localhost/FullPageDashboard do Not load. Is there a way to start these (maybe a service?) via shell script?

guysoft, 2016-11-13:
This is known, it requires to change how the custom session starts on
pixel. Again, help would be great because I am busy with my startup company.

guysoft, 2016-12-25:
New image ( 2016-11-25) was released, and has this written as a change: ``Updates to hardware video acceleration in Chromium browser``

http://downloads.raspberrypi.org/raspbian/release_notes.txt

Will build a nightly with it so we can see if it changes anything

commit 342ab281 (2016-12-27): Add GPU acceleration, fixes #75 #29 #22
closed 2016-12-27 by guysoft