guysoft/FullPageOS #76: Disable Chromium --incognito

Jamy, 2016-10-12:
This isn't really an issue, per se, but I have a website application that has an auto login feature that won't work with incognito more.

What file can I edit to make Chromium start without Incognito mode?

Thank you.

spencerlandis, 2016-10-13:
I'd like this option as well if it persists cookies between boots/sessions.

guysoft, 2016-10-20:
Yes you can, edit the line ``home/pi/scripts/start_chromium_browser``

The source for comparison is here:
https://github.com/guysoft/FullPageOS/blob/devel/src/filesystem/home/pi/scripts/start_chromium_browser

You can append any flag you want to the chromium call there

Jamy, 2016-11-27:
I forgot to write back some time ago, but I did notice chromium does not start with incognito mode as default, so it worked well for me.

However, it is very helpful to know the location of the startup script, so thank you for that.