guysoft/FullPageOS #79: Hide boot command text

Sn0ken, 2016-11-03:
I am using Fullpageos 201609-23 0.7.0 to display a webpage with realtime production information. I would like to hide all boot command text during boot. I also want to hide the raspberries icons and the color test splash.
Then I wanted to add a splash logo before the webpage loads. This will be a company logo.

I used this guides to add a customer splash
https://github.com/elalemanyo/raspberry-pi-kiosk-screen
http://raspberrypi.stackexchange.com/questions/1214/how-to-add-custom-loading-screen

Then I added this line to config.txt to remove the colored square
disable_splash=1

I added this lines to cmdline.txt to hide the berries and to hide boot command text.
logo.nologo
loglevel=3
quiet
and changed console=tty1 to console=tty3
vt.global_cursor_default=0

The custom splash works, the berries is hidden and the colored square is hidden. The issue is that the boot command text is still there.
Every guide that I find in the internet tells me to change tty1 to tty3 to hide the boot command text. I have tried it in Raspbian and it works.
Can someone give me some idea how to remove preferably all boot text?

strabbit, 2016-12-04:
I ran into the same issue... `console=tty42` did the trick for me :)

Sn0ken, 2016-12-05:
Thanks for the tip! I will try this.

cesarzapatawfm, 2017-01-05:
I'm trying to do the same but when I run this apt-get install fbi , is failing to fetch 
E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/g/ghost                                                            script/libgs9_9.06~dfsg-2+deb8u1_armhf.deb  404  Not Found [IP: 5.153.225.207 80                                                            ]

E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/g/ghost                                                            script/ghostscript_9.06~dfsg-2+deb8u1_armhf.deb  404  Not Found [IP: 5.153.225.2                                                            07 80]

any ideas?

guysoft, 2017-01-05:
``sudo apt-get update``

cesarzapatawfm, 2017-01-09:
thanks. the sudo 'ap-get update' works to get fbi working. However, after following the links supply by OP i'm still getting the default command text and berries. no splash screen at all. 

if I run 'sudo /usr/bin/fbi -T 1 -noverbose -a /etc/splash.png' I can see the custom image. So I know fbi and the .png are working. 

has anyone successfully added the a custom splash screen to fullpageos? 

thanks

Sn0ken, 2017-01-10:
For us fbi works great. It shows our company logo a couple of seconds.

mrozendaal, 2017-03-03:
put "sleep 5" without the quotes in the script for showing the splashscreen foor 5 seconds