guysoft/FullPageOS #81: Multiple FullPageOS Pis on the same network causes Chromium profile issue

ryanregall, 2016-11-08:
I get an error hen I have more than one FullPageOS Pi running on the same network: "This Chromium profile is in use on another computer on the network."  I press the button to "unlock profile and relaunch chromium" and that seems to fix it, even after rebooting that error message doesn't re-appear.

Is this a long-term fix, or do i need to change something else in the chromium settings to be able to have multiple FullePageOS Pis on the same network?  Do I just "unlock profile and re-launch" for each Pi I add, and that is the fix?

Thanks for making FullPageOS.  It's exactly what I was looking for.

fazen, 2016-11-08:
Same problem here when I changed my hostname following directions of #71

fazen, 2016-11-25:
You can reset the Chromium session also from the shell, with this command:

`rm -rf /home/pi/.config/chromium/Singleton*`

Useful if you can't interact with the UI or if you want to fix with a script.

guysoft, 2018-11-21:
Is this still relevant?

patricktokeeffe, 2019-01-09:
Hi, just confirming OP's symptoms do still occur when the hostname is changed. Using version 2018-04-22 on a Pi2. Plug in a keyboard and press 'space' to unlock & relaunch.

> "This Chromium profile is in use on another computer on the network." I press the button to "unlock profile and relaunch chromium" and that seems to fix it, even after rebooting that error message doesn't re-appear.

guysoft, 2019-01-10:
Ok. found:
https://bugs.chromium.org/p/chromium/issues/detail?id=367048
Which claims to have been fixed.
But is mentioned here:
https://askubuntu.com/questions/476918/google-chrome-wont-start-after-changing-hostname/476932

I can add this line recommended to the build, and see if it fixes that:
```
rm -rf /home/pi/.config/chromium/Singleton* 
```

guysoft, 2019-01-10:
Ah wait, I get it,
The how are you people changing the hostname to differentiate between the machines?

You need to run that command after changing the hostname and booting, if I add it to the build it would make no difference.

patricktokeeffe, 2019-01-10:
> if I add it to the build it would make no difference.

Yes I agree because it results from the hostname update. I was able to fix this with a simple GUI acknowledgement and the fixes are persistent. There's enough info here it's easy to work around. Thanks everyone