guysoft/FullPageOS #82: Login

albertvveld, 2016-11-14:
is it possible to auto login on websites?

guysoft, 2016-11-16:
Same as chrome browser. You can install an extension to do that, for example:
https://chrome.google.com/webstore/detail/auto-login/kjdgohfkopafhjmmlbojhaabfpndllgk?hl=en

monotek, 2017-04-04:
Why not us xdotool as its installed anyway?

Example to do a google search:

```
#!/bin/bash

PAGE_URL="https://www.google.com"
PAGE_FORM1="Fullpageos"

xdotool windowactivate --sync $(xdotool search --onlyvisible --name chromium)
xdotool key CTRL+l
xdotool type ${PAGE_URL}
xdotool key Return
sleep 1
xdotool key CTRL+a
xdotool key Delete
xdotool type ${PAGE_FORM1}
xdotool key Return
```

guysoft, 2017-04-04:
You can also use chromium to save logins (open a window with ctrl-t).

Also any chrome extension that saves passwordsM

monotek, 2017-04-04:
Yes, but it doesn't feels save to trust some chrome app and give them my passwords.

guysoft, 2017-04-04:
Well, i can't build anything beyond that myself, and you are using my
distro (unless you built it yourself).

guysoft, 2017-09-26:
Closing if no reply

closed 2019-08-29 by guysoft

Norfeldt, 2022-05-02:
(I know that the thread is old, but just in case this could be of any help)

If it's an [Chromium login issue](https://forums.raspberrypi.com/viewtopic.php?t=299392) (which I had a lot of struggle with) then try:

`Chromium settings`->`Advanced`->Turn **off** `Allow Chromium sign-in.`

For other logins and if you don't wanna use some chrome extensions. Then you might be able to use [chromes puppeteer recorder](https://developer.chrome.com/docs/devtools/recorder/) and see of you can trigger that to run in chromium,

guysoft, 2022-05-02:
Might be worth adding that in to the wiki on this repo