guysoft/FullPageOS #83: CSS3 Not working

mflmartin, 2016-11-16:
I have a pretty simple page, as a matter of fact I tried this one:

https://codepen.io/davidhc/pen/nLpJk/

(just export it, put it online and load it in rapsberry pi 2 with FullPageOS)

Nothing is on display. If I just put regular text, it loads it correctly. And regular images as well. But the moment I put something with CSS3, or keyframes, etc, it doesn't work.

mflmartin, 2016-11-16:
Update:

The problem was not CSS, at all.

My mistake. The problem where JPG files. Some JPG Files do not show under chromium (god knows why, because in chrome, firefox, et al, they work).

I attatched 2 images, 

if you add them (with an img tag) to a simple html file, they don't work.

Weird as hell :S

Maybe something to do with the encoding of the JPG?

fazen, 2016-11-16:
It's discussed here: #66

guysoft, 2016-11-17:
Ok, closing and you can continue talks in #66 . Not to confuse people that CSS and let them know CSS is working.

closed 2016-11-17 by guysoft

mflmartin, 2016-11-17:
Agree. Sorry for the misleading title, due to my mistake inspecting the issue. I updated the title. 

Thanks!