guysoft/FullPageOS #84: Vertical mode?

mansj, 2016-11-30:
I'd like to run my information dashboard screens mounted vertically, i.e. rotated 90 degrees. Is this possible with FullPageOS somehow?

guysoft, 2016-12-01:
edit:

    sudo nano /boot/config.txt

Add one of these to the bottom:

    display_rotate=0 Normal
    display_rotate=1 90 degrees
    display_rotate=2 180 degrees
    NOTE: You can rotate both the image and touch interface 180º by entering lcd_rotate=2 instead
    display_rotate=3 270 degrees
    display_rotate=0x10000 horizontal flip
    display_rotate=0x20000 vertical flip

source:

https://www.raspberrypi.org/forums/viewtopic.php?f=108&t=120793

closed 2016-12-01 by guysoft

Morphy99, 2017-07-21:
I've tried this with display_rotate=2 and also lcd_rotate=2 and neither affect my attached rpi 7" touchscreen display. It does work on other distros however and I've noticed that the rainbow pattern is rotated but not the FullPageDashboard.

guysoft, 2017-07-23:
@Morphy99 , it won't work, the RaspberryPi kernel team has pushed a fix see https://github.com/guysoft/FullPageOS/issues/137

JMurph2015, 2019-02-19:
Sorry to necrobump this, but using "display_rotate" requires disabling the OpenGL driver, which is a big performance issue for Chrome/Chromium.  I've heard that one can use xrandr to achieve the same thing, but it's unclear what needs to be done to make that happen.  Thoughts?

guysoft, 2019-02-21:
I have no time to test, so no input here.
Indeed OpenGL makes things faster, I think xrandr uses the display driver which does not support it.
Also the upcoming RC will have a "non acceleration version" so you could flash that and rotate it right away.