guysoft/FullPageOS #86: Awww snap

LOFer, 2016-12-14:
I am running this on our school reception TV and it has been going fine for a month or so but recently, a few times a day we now get the white screen of doom with the "Aww Snap" message.

The web server is running fine and network connectivity is not a problem as I can connect from another machine so it is something to do with the local (to the Pi) instance of Chromium that is running.

Anyone had this before - how can I fix this? Is it a cache issue and if so, where is the cache stored for this?

Cheers!

0xE1, 2016-12-14:
"Aw, Snap!" have multiple error codes, which one are you getting?
[https://support.google.com/chrome/answer/95669](https://support.google.com/chrome/answer/95669)

All of them are related to network in one way or the other.

guysoft, 2016-12-27:
The cache is sorted in:
```
/home/pi/.config/chromium/Default/Application Cache
/home/pi/.config/chromium/Default/Storage/ext/chrome-signin/def/Cache
/home/pi/.config/chromium/Default/Storage/ext/chrome-signin/def/GPUCache
/home/pi/.config/chromium/Default/GPUCache
/home/pi/.config/chromium/Default/History Provider Cache
/home/pi/.config/chromium/Default/Service Worker/CacheStorage
/home/pi/.config/chromium/Default/Service Worker/ScriptCache
/home/pi/.config/chromium/ShaderCache
/home/pi/.config/chromium/ShaderCache/GPUCache
```

pablosouza10, 2016-12-29:
Same problem!

Using new FullPageOS RC1 and when run Youtube Video  "Aw Snap"

Any fix?

guysoft, 2016-12-30:
@pablosouza10 that's a different issue, see #29 webm and flash videos should work. But mp4 is a problem. Also upvote this :) http://raspberrypi.stackexchange.com/questions/59524/chromium-tab-crashes-if-mp4-video-with-gpu-acceleration

pablosouza10, 2016-12-30:
@guysoft , Tks. Now I'm try New Raspbian Pixel, Youtube's hd videos running so much better w/ no modification.  Waiting for fix FullPageOs.

guysoft, 2017-06-30:
@pablosouza10 Ok I think I understand what they did in Pixel.
Pixel comes default WITHOUT GPU accleration. 
However, it comes per-installed with the [h264ify extension](https://github.com/erkserkserks/h264ify). 
So you can either have GPU enabled and have accelerated canvas. Or disable the GPU and have h264ify decode youtube videos. Also see https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=157049.

guysoft, 2017-08-11:
Closing if no reply

tailorvj, 2017-08-14:
@guysoft Raspberry Pi forum lists a few workarounds for the accelaration problem. Do you think any of them can temporarily solve the problem of accelarated video playback until this issue is taken care of by Raspbian?

clear_uniforms_before_first_program_use (I guess this is a Chromium flag?)

Cheers,
Tailor

guysoft, 2017-08-14:
Can you please link to a specific solution? I gave q pretty solid one two comments back.