guysoft/FullPageOS #87: Few questions about features

jsparacino, 2016-12-15:
I got everything to work fine but we may have a need to display one page, wait a few seconds, then display a different page.  Is there a way to do that?

We are also looking into a way to be able to "deploy" a fullpageos.txt change to a different page.  We might have multiple displays and we made need to change what they are viewing.  We want to do that remotely.  So I was wondering if there is a way to have chromium see that URL change in the fullpageos.txt and refresh the display with the new URL without having to reboot or have someone walk over there and refresh it?

Then, maybe I just missed it, but how do we access the FullPageDashboard?

Thanks for any help!

guysoft, 2016-12-16:
1. Yes, that is what FullPageDashboard does, its set to 30 seconds.
2. Yes, you can either modify the SD before flashing it, using mount_image function in the sources, or use something like [fabric](http://www.fabfile.org/) to ssh to the PIs and modify them when they are already running. You can then rung a script that refreces the pi in the`` ~/scripts`` folder
3. You access FullPageDashboard using ``http://your_pi_address/FullPageDashboard``
4. It would be great if you can document what you are doing FullPageDashboard is no manitained anymore by its developer and help is appreciated, @tailorvj is also tracking that.

jsparacino, 2016-12-16:
Which Image do you recommend we use?  I am having a hard time getting the IP set to something static that we are setting.  I tried changing the fullpageos-network.txt and I tried changing modifying /etc/network/interfaces with no luck keeps pulling DHCP. I am using 2016-07-20_2016-05-27-fullpageos-jessie-lite-0.7.0

jsparacino, 2016-12-16:
For the image I am using I discovered that dhcpcd was overriding the scripts and /etc/network/interfaces

guysoft, 2017-08-11:
For static dhcp use now https://github.com/guysoft/OctoPi/issues/219

closed 2017-08-11 by guysoft