guysoft/FullPageOS #91: Chromium does not play mjpeg

graffmaxx, 2016-12-22:
Hi there,

Does anyone know what packages are needed to enable mjpeg webcam streams on chromium?
Currently i use an uv4l webrtc streaming server. If i open the adress on my notebook or tablet everything works fine. Also if i use a standard raspbian pixel image. But not if i use fullpageos. On this image i only see a white screen with a broken jpg symbol.

Thanks for help

graffmaxx, 2016-12-22:
Forgot to say that it is a IP Cam

guysoft, 2016-12-27:
Do you have a video example I can test on 0.7.0 RC1?
see #92

graffmaxx, 2016-12-28:
Hi there,

i used the image "2016-10-20_2016-09-23-fullpageos-jessie-lite-0.7.0"
On this image i couldn't display any mjpeg stream in chromium.
You can try this URL. This one also didn't work: 

**http://www.ipcams.ch/WebCam.aspx?nr=856**

But now i fixed the issue by setting up my own Kiosk Image with your scripts and a original raspbian jessie pixel. Now everything works. 

I used your chroot_script and copy pasted every command expect the apt-get remove commands.

Maybe you remove to many libraries in the installation scripts?

guysoft, 2016-12-28:
Hey, I just tested it on RC1 nightly and it works, use that.

closed 2016-12-28 by guysoft