guysoft/FullPageOS #92: FullPageOS 0.7.0 RC1

guysoft, 2016-12-27:
FullPageOS 0.7 first release candidate.
The main feature added is GPU acceleration.

Features:
* Add GPU acceleration, fixes #75 #29 #22 
* Added Adobe flash
* New Raspbian 2016-11-25
* Using native chromium used by Pixel
* Lots of new configuration options, such as timezone #63 password and locale #64 - Thanks @semanticfire
* Better touch support #78 - thanks @NoviCMS 

Please try it out and comment if there is any problem, before this becomes the main version:

http://docstech.net/FullPageOS/nightly/2016-12-27_2016-11-25-fullpageos-jessie-lite-0.7.0_RC1.zip
md5: 
http://docstech.net/FullPageOS/nightly/2016-12-27_2016-11-25-fullpageos-jessie-lite-0.7.0_RC1.zip.md5

P.S, you can test that GPU acceleration is working by pressing ``ctrl+tab``, and going to the url ``chrome://gpu``. This is what you should see. Tips on how to improve this would also be appreciated!
[image]

guysoft, 2016-12-27:
Possible bug on video playback: https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=168560&p=1087990#p1087990

guysoft, 2016-12-28:
Stuck, upvotes appreciated :) : 
http://raspberrypi.stackexchange.com/questions/59524/chromium-tab-crashes-if-mp4-video-with-gpu-acceleration

mansj, 2017-03-03:
On my brand new installation using the RC 1 img linked in this thread, I get "hardware acceleration disabled" on _all_ of the points under "Graphics feature status". 

I'm using a brand new Raspberry Pi 3. 

[image]

guysoft, 2017-03-06:
@mansj Strange, I have no idea why it would work on one and not the other. What make is the hardware?

guysoft, 2017-03-31:
Closing, use RC2
https://github.com/guysoft/FullPageOS/issues/92

closed 2017-03-31 by guysoft

guysoft, 2017-04-04:
RC2 is broken, use RC1 until RC3 comes out