guysoft/FullPageOS #94: Chromium won't load certain pages

407pilot, 2017-01-01:
I'm trying to use FullPageOS to act as a weather display for a helicopter EMS base, but for some reason Chromium won't load [this page](http://www.aviationweather.gov/hemst). Yesterday, it worked once. I'm not at all proficient with Raspbian, so any help is greatly appreciated.

guysoft, 2017-01-02:
1. That page has a statment you need to approve.
Ceiling and Visibility Precautionary Use Statement

This product is for flight planning purposes only and should always be used
in combination with ceiling and visibility

2. What do you see when it "does not work"?

407pilot, 2017-01-02:
1. The agreement page won't load either, even if I point it to the [main link](http://www.aviationweather.gov/hemst/) instead of the permalink.

2. It's just a blank white screen. Nothing loads at all. Some other pages work fine ([like Weathermeister](http://www.weathermeister.com))

407pilot, 2017-01-02:
Sorry about the bad picture. I can't VNC into the Pi for a screen shot in the network it's currently on.

[image]

pwillard, 2017-01-12:
I have some similar issues... I get my page working... claim success and move it to its final location upstairs and bam... page never displays right again.  It's nuts.  (page works fin on Chrome in the PC)

tailorvj, 2017-02-28:
FullPageDashboard has all kinds of funny behaviours and we intend to retire it. We are going to need some help from JavaScript programmers to do it though, so please feel free to contact me if you would like to invest a few hours of coding into this.

guysoft, 2017-03-05:
Unless anyone wants this open, I am closing, its a FullPageDashbaord issue and there is an  issue tracker there.

closed 2017-03-05 by guysoft

1am, 2017-08-19:
@tailorvj Regarding [your last post in this thread](https://github.com/guysoft/FullPageOS/issues/94#issuecomment-282986730) it can actually be simplified and also resolve the issue with embedding https pages in iframe which I've found to be a problem with [FullPageDashboard](https://github.com/amitdar/FullPageDashboard) (though I've resolved it with [Ignore X-Frame headers plugin](https://chrome.google.com/webstore/detail/ignore-x-frame-headers/gleekbfjekiniecknbkamfmkohkpodhe) Not a good way to handle but it worked for the time being.)

The solution I've eventually ended up with is installing [Revolver Tabs chrome plugin](https://chrome.google.com/webstore/detail/revolver-tabs/dlknooajieciikpedpldejhhijacnbda) to the browser, set it up and modify the **scripts/start_chromium_browser** script to:

```
chromium-browser \
        --kiosk \
        --touch-events=enabled \
        --disable-pinch \
        --noerrdialogs \
        --disable-session-crashed-bubble \
        https://www.ixquick.com \
        https://google.com \
        http://www.bing.com
```

This could probably be automated in some way to build ready images but that part was out of scope of my work.

guysoft, 2017-08-19:
I can't find a way to automate extension installation. I also wanted to
install h264ify. Ideas welcome.

1am, 2017-08-20:
Since the extensions are already in Chrome Web Store the 2nd point from [Stack Overflow - How install crx Chrome extension via command line?](https://stackoverflow.com/a/34178700) and the link provided there might be a good reference.

guysoft, 2017-08-20:
Did you mean to link that or the part about ``unpacked extensions`` listed in the second answer?
Will note the raspberrypi foundation do ship with extensions installed, not sure how they do it, does not appear in pi-gen.