guysoft/FullPageOS #96: FullPageOs won't boot

MRine00, 2017-01-12:
I did everything as said in the readme file. Yet my FullPageOS won't boot.

When i start my Pi, it shows the colored pi boot screen. And it's just stuck. Nothing happens.

What am i doing wrong?

guysoft, 2017-01-14:
1. What pi? 
2. Are you using a 2A power supply with a GOOD usb cable?

jorgeiglesiasca, 2017-01-26:
Same to me with latest RC1.   Tried with many 2Amps chargers but none of them seems to fix.

guysoft, 2017-01-27:
What raspberrypi version?

jorgeiglesiasca, 2017-01-27:
Tried with 2 and 3 pi

guysoft, 2017-01-27:
Does it work with 0.6.0?

jorgeiglesiasca, 2017-01-27:
Yes.  Worlking with 0.6 for months.  Allways pi2. Tried 0.7  boot screen
but no signar alfter. Tried pi3 and the same.   Changed chargers with one
up to 2.4a and nothing changes.

Just in case helps im always using hdmi to vga adapter an the screen says
resolution not supported. Modified resolution in config to 1024 x 768 to
test and no difference.

Lcd is fullhd

jorgeiglesiasca, 2017-02-03:
I have made some tests.  The problem happens only with the lite isos , not with the full size ones , the full size iso boots to desktop but fullpageos.  At least it works .

guysoft, 2017-02-03:
@jorgeiglesiasca  Are you using 0.70RC1 or just 0.7.0?

durango, 2017-02-20:
Running into the same issue actually on pi 2 with the full version (RC1)

guysoft, 2017-08-11:
Closing due to no reply

closed 2017-08-11 by guysoft

Teku16, 2018-06-11:
I am having this issue. 2018-04-18-fullpageos-stretch-lite-0.9.0.img is the filename.