guysoft/FullPageOS #98: Blank page after days of running fine. did Chrome crash?

cesarzapatawfm, 2017-01-22:
I have about 5 RPis running fullpageos to display custome made dashboards. Everything is working fine but couple of time so far, chrome has been showing a blank page only.

I'm able to vnc to the RPI3 in question and just doing a F5 fixes the issue but I wonder if there is a crash in the browser or somewhere else. I'm unable to do F12 (developers tools ) to see if there is any client error. 

What is the way to troubleshoot this? is it fullpageos, chrome or my app that is failing?

guysoft, 2017-01-22:
What version of FullPageOS?
I am getting it in 0.7.0 RC1

cesarzapatawfm, 2017-01-24:
2016-05-10-fullpageos-jessie-lite-0.6.0.img I'm using this. I'm not sure how to get the version number.

guysoft, 2017-01-25:
Thats 0.6.0. Sounds like a browser crash, the way to check it is using chrome debug console: https://developers.google.com/web/tools/chrome-devtools/console/

cesarzapatawfm, 2017-02-27:
Do you do that with Chromium on FullOS? I tried pressing F12 or going thru the menu but I can't get the developer tools window to show up.

guysoft, 2017-09-24:
Closing if no activity, chromium getting much more stable now.