guysoft/FullPageOS #99: Cant seem to get any website to load, except basic google search page.

ogbobby, 2017-01-25:
i am using this version:2016-07-20_2016-05-27-fullpageos-jessie-lite-0.7.0.zip.
After config the wifi and adding just a basic page to fullpageos.txt, upon reboot it just boots to a google search page. Am i missing something?

guysoft, 2017-01-25:
Can you paste your``fullpageos.txt``?

ogbobby, 2017-01-25:
it just contains one url at the top:
http://www.runthejewels.com

This isnt the actually link i will be using it was just one i was testing.

guysoft, 2017-01-25:
Ok, can you pate the output of:
``head -n 1 /boot/fullpageos.txt`` ?

ogbobby, 2017-01-25:
http://www.runthejewels.com

guysoft, 2017-01-25:
Can you really paste it? as in with the terminal output? I want to debug this.

ogbobby, 2017-01-25:
im not sure how to do that when using putty ssh'd into the pi, can you tell me how. I apologize this is all pretty new to me.

ogbobby, 2017-01-25:
Ok nevermind i think i got it. Thank you for helping and for your quick response and for all the work you have done to create this!!!!

closed 2017-01-25 by ogbobby

guysoft, 2017-01-26:
What was wrong?

ogbobby, 2017-01-26:
It was my own dumb fault